ohAnd/dtuGateway #6: Error Build in platform.io - buildnumber file not found

carsten-wolf, 2024-03-26:
If I try to build the Version (main oder snapshot) the build failed with this error:

```
Processing esp12e (platform: espressif8266; board: esp07s; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
get current version: 1.4.0
 from C:\Users\User\Downloads\dtuGateway-snapshot/include/version.h
buildnumber file not found
NameError: name 'build_string' is not defined:
  File "C:\Users\User\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 167:
    env.SConscript(env.GetExtraScripts("pre"), exports="env")
  File "C:\Users\User\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\User\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\User\Downloads\dtuGateway-snapshot\version_inc.py", line 40:
    if build_string != "localDev":
```
Has someone an idea what is wrong?

ohAnd, 2024-03-27:
Hi,
is a small hack to have easy remote versioning with github actions. For local run it's the following needed:

'ERROR: buildnumber file not found. For automatic versioning there is a file called ../include/buildnumber.txt expected. With the content "localDev" or versionnumber e.g. "1.0.0" in first line. (File is blocked by .gitignore for GitHub actions to run.)'

I've added this info to the error message. (in develop branch)

carsten-wolf, 2024-03-28:
worked for me with adding this build file and localDEV.
Thanks for the support!

closed 2024-03-28 by ohAnd