ohAnd/dtuGateway #11: Mqtt?

Lassa333, 2024-04-28:
Hi,
First of all: massive Respekt for your great Job!!!

For different reasons I can not Deal with openHub. But with Mqtt.

Would it be a Bug Deal to integrate in your next Release the possibilty to parse the data as mqtt? 

This would open your project to mich more perplex

BR
Roland

Kscho81, 2024-04-30:
Sehr gute arbeit. Wenn ich nun noch die Daten in den IObroker bekomme, hätte ich alles was ich brauche. Zur Zeit geht es ja nur mit OpenHab. Ginge es nicht auch generell mit einem MQTT Broker ?

ohAnd, 2024-05-09:
hi,

I will check that. Should be no big deal ...
But in my environment I have no productive use of a MQTT broker... (for dev I will check with a small Mosquitto inctance) So there is no useful long time testing possible.
Therefore it will be helpful, if you can do some testing on your side...

Let's give it a try.

Lassa333, 2024-05-09:
Hi Andreas,

(Ich bin aus Deutschland, also können wir auch deutsch schreiben ;) )

I like your setup. As I own a HMS800W2T I can not use an opendtu anymore. But I am not able to follow your reverse engineering - which I would like to do. So I have only Little ways to keep my Inverter values (without a China cloud)

So for me (and I think many others, too) it would be great to Extrakt the values of the Inverter and "handle them over" in an open/basic way: MQTT

With the Inverter values extracted and published over MQTT, one is free to do whatever is wanted:

- Integration into Homeassistant, OpenHab, and any other smarthome Systems you can think of
- collect the values in an influxDB and visualize them with Grafana
- build automations based on the Inverter values
- ...

As you can See, there would be only a Little step to go for you and So many guys could be absolute happy with the New flagship of Hoymiles (hms800w2t) and your brillant work on the retro-engeneering. Wouldnt it be worth it?

BR
Roland 

> hi,
>
> I will check that. Should be no big deal ...
> But in my environment I have no productive use of a MQTT broker... (for dev I will check with a small Mosquitto inctance) So there is no useful long [...]

-------------------------------------------------------------------------------------------------
FreeMail powered by mail.de - MEHR SICHERHEIT, SERIOSITÄT UND KOMFORT

Lassa333, 2024-05-09:
Hi Andreas,

I have no esp32 device, yet. But I can order one and  I could run a mosquitto server on one of mine rasperberrys, too

would love to do the testing of your step forward :)

Just give me a word

Lg
Roland

> hi,
>
> I will check that. Should be no big deal ...
> But in my environment I have no productive use of a MQTT broker... (for dev I will check with a small Mosquitto inctance) So there is no useful long [...]

-------------------------------------------------------------------------------------------------
FreeMail powered by mail.de - MEHR SICHERHEIT, SERIOSITÄT UND KOMFORT

ohAnd, 2024-05-09:
Hi Roland,

I've implemented the first "try" ... see feature branch.
Now it is possible to configure a MQTT broker with user / pass and main topic for the published data of the dtuGateway.
Also the two "bindings" openhab and mqtt are activatable in parallel or each one alone.
The settings UI is adapted accordingly.

In MQTT Explorer it looks like this:
[image]

next step: subscribing for a topic on the mqtt broker to get the power limit set by an external source to submit to the dtu (similar with the current openhab solution)

Maybe you or someone can give a try...

BR

Lassa333, 2024-05-10:
Hi Andreas,

SUPER!!!!

I ordered a esp32 (2866) yesterday. I should get it on Monday or Tuesday. Than I will dig into flashing an esp32 device. Maybe I can rig up the MQTT test device this weekend.

Do you have any specific questions to be answered after the first test period? (E.g. stability, loss of connection, workability, plausabillity,....)

BR
Roland 

>
>
> Hi Roland,
> 
> I've implemented the first "try" ... see feature branch.
> Now it is possible to configure a MQTT broker with user / pass and main topic for the published data of the dtuGateway.
> Also the two "bindings" openhab and mqtt are activatable in parallel or each one alone.
> The settings UI is adapted accordingly.
> 
> In MQTT Explorer it looks like this:
> image.png (view on web)
> 
> next step: subscribing for a topic on the mqtt broker to get the power limit set by an external source to submit to the dtu (similar with the current openhab solution)
> 
> Maybe you or someone can give a try...
> 
> BR

-------------------------------------------------------------------------------------------------
FreeMail powered by mail.de - MEHR SICHERHEIT, SERIOSITÄT UND KOMFORT

ohAnd, 2024-05-10:
Hi Roland,

sounds good.
At this point I realized I've not mentioned in the readme for which specicfic SOC this project is build...

=> I've used ESP8266 based baords for that e.g. [this ](https://www.amazon.de/gp/product/B0754W6Z2F/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) or [similar ones](https://www.amazon.com/s?k=esp8266&__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&ref=nb_sb_noss).

A porting for pure ESP32 is not efficient at this stage, so be sure you will get the right one. 

For this project there are no ports/ GPIOs used, so the board can be the "smallest" one based on an ESP8266 MCU.

For testing - as you already mentioned -> using in productive environment (Hoymiles inverter with MQTT broker e.g. IObroker), checking for usability, stability - finding errors...

BR

Lassa333, 2024-05-10:
Hi,

I screend your Main.cpp and found, that you wrote the Code for a 2688 esp Board. So I bought a wemos D1 Mini Pro. This should be a 2688 chip with 16MB memory. I hope, this Board will work. 
I added the description of the Board i bought. If this is not the right one, please let me know.

I will build up a productive System (ip 192.168.0.100 should be the right one?) With a MQTT server. This server should receive the messages of your gateway extracted out of my hoymiles hms200w2t Inverter. 

As soon my System is up and working, I will keep you informed

BR

> Hi Roland,
>
> sounds good.
> At this point I realized I've not mentioned in the readme for which specicfic SOC this project is build...
>
> => I've used ESP8266 based baords for that e.g. this or similar [...]

-------------------------------------------------------------------------------------------------
FreeMail powered by mail.de - MEHR SICHERHEIT, SERIOSITÄT UND KOMFORT

Lassa333, 2024-05-16:
Hi,
I miss a tab for the mqtt settings. 

Where can I Insert the IP of the mqtt broker, username and password?

Br

LukeTheMast3r, 2024-05-17:
Hi,
cooles Teil Funktioniert auch soweit. 
Die MQTT integration würde ich auch gerne testen. 

@ohAnd könntest du das als Release  oder so bereitstellen? Also eine .bin Datei?
Bin leider noch nicht soweit um das selbst zu erstellen.

Grüße

Hi,
cool part works well so far. 
I would also like to test the MQTT integration. 

@ohAnd could you provide this as a release or something? As a .bin file?
Unfortunately I am not yet ready to create it myself.

Regards

LukeTheMast3r, 2024-05-17:
@Lassa333 aktuell fw version: 1.5.0028

Lassa333, 2024-05-17:
Hi,

 

Ich nutze auch die Version 1.5.0028

 

Aber ich finde die Einstellungs-Seite nicht, um mqtt einzustellen. Mein mqtt-Broker läuft schon. Und langweilt sich und wartet auf Werte im Topic

 

lets stick to English 😊

 

I also use the Version 1.5.0028 

But I am not able to set the settings for the mqtt-Connection in the firmware. Did you find the settings-page? My mqtt-Broker is up an running. And waits for some Topic-values

 

BR

LukeTheMast3r, 2024-05-17:
Hi Roland,

you cannot find the mqtt settings in this version.
There is a third branch "11-mqtt" besides main and develop.

Think you have to donwload the sourcecode, compile it yourself to an .bin file and flash the ESP.
Unfortunately i am not (yet) able to do this. So we have to wait for Andreas to do so.

Could you please delete the copied "reply" Text in your email Answer in the further conversation? 
Would help to keep this issue a bit smaller :)

Lukas

Lassa333, 2024-05-17:
Hi,

OK, I will improve:)

I think we are Stuck on the same step. I have no experience in compiling a Firmware. I can read the code/Sketch but I understand not even the half :/

So I think we have to wait for Andreas

Br  

-------------------------------------------------------------------------------------------------
FreeMail powered by mail.de - MEHR SICHERHEIT, SERIOSITÄT UND KOMFORT

ohAnd, 2024-05-18:
Hi,
to make it easy/ fast I've merged the stuff from "11 mqtt" to dev, so the build will be directly available as the [latest snapshot](https://github.com/ohAnd/dtuGateway/releases/tag/snapshot).

If you can spend some time for testing, it will be nice.

Thx

LukeTheMast3r, 2024-05-18:
Thanks alot.

Have updated to the new version. Cannot etablish a connection. The reason is that i use a custom Port for mqtt.
If i type in 192.168.2.144:11883 the Gateway cut's it to this: 192.168.2.144:1

Could you add an Field for the Port or simply recognize the port if there is something after the ":" like this:

192.168.2.144:11883

?

Thx

ohAnd, 2024-05-18:
Hi Lukas,

currently there is only the standard mqtt port supported see [code](https://github.com/ohAnd/dtuGateway/blob/c1902dfafa2869d3c99a0258f30a8d23d774f14c/src/dtuGateway.ino#L868)

=> 1883

But it make sense to bring this also to the user settings.
Give me some time.

BR

Kscho81, 2024-05-18:
[image]
... hallo zusammen, mit der neuen version bekomme ich keine daten.

ohAnd, 2024-05-18:
@Kscho81 can you please check your settings...
As described before there are breakable changes in user settings and you have to reconfigure all local data incl. dtu connection IP.

In my environment the fw 1.5.0030 running well.

Otherwise there is some more info needed to find the problem. (e.g. DTU is up and running, reachable inside your network, ...)
If you connect your PCB with a COM terminal you can see debug mesages and states to find out...

Lassa333, 2024-05-18:
Hi,

For me the version 1.5.0030 works very good!

I got connection to the inverter (when there was sun) just as with the older versions

I was able to set up the mqtt-connection (with port 1883) and I received data on the set topic. They were updated ~every 30 seconds (as expected). The data are stored in my test-environment in a InfluxDB

So fare – very good!!! Great job, Andreas!!!

One minor thing: in the frontend is a wrong version-number displayed (1.5.0029) 😊

ohAnd, 2024-05-18:
Hi together,

I've integrated the variable port setting for the broker connection. see [latest snapshot](https://github.com/ohAnd/dtuGateway/releases/tag/snapshot).

With this point I will close this issue - basic integration of MQTT is done.

For possible still open side topics or new feature requests/ discussion, please raise dedicated new issues, to discuss specific solutions.

(@Kscho81 maybe regarding your connection problem or @Lassa333 for the topic with the wrong firmware (I can't reproduce on my side)

thanks

closed 2024-05-18 by ohAnd

Lassa333, 2024-05-19:
Hi Andreas,

 

While contributing to your brilliant work on your dtuGateway, I realised that your gateway is not fare away from existing dtu (e.g. openDTU or AhoyDTU) – but with the WIFI-component. So after the slick adding of the MQTT-functionality, I want to ask you a question:

 

Are you interested in setting up a “WIFI-dtu”?

 

I would love to hear your thoughts about this

 

BR

Roland

Lassa333, 2024-05-21:
Hi Andreas,

I just wanted to let you know, the FW 1.5.0031 works as supposed.

But I realized, the there are many inverter reboots (109 in the last 2 days)

Is this a problem? Are you aware of this?

BR

Lassa333, 2024-05-21:
Hi 
Klappt es nun?
Lg

-------------------------------------------------------------------------------------------------
FreeMail powered by mail.de - MEHR SICHERHEIT, SERIOSITÄT UND KOMFORT

Kscho81, 2024-05-21:
1.5.0034 läuft stabil- DTU reboots (1x pro Tag). MQTT läuft auch sauber im IObroker. Sehr gute Arbeit. Vielen Dank

Lassa333, 2024-05-24:
Hi Andreas,

it would be great to add an OLED-Display to show crucial values out of the inverter. With this your gateway is totally suitable for the new Hoymiles-Flagship. And a perfect replacement for OpenDTU/AhoyDTU

I could provide a picture of the values a ahoyDTU shows on the display

BR