ohAnd/dtuGateway #14: settings lost after update -> restore settings

LukeTheMast3r, 2024-05-18:
Hi, 

everytime i Update the Gateway the whole settings are lost.

Is it possible to safe them before the update and restore them afterwards?

Or am i doing something wrong?
I'm using the onboard ota update process.

Thanks

Greetings
Lukas

ohAnd, 2024-05-18:
Hi Lukas,

all settings will be stored in EEPROM (8266 emulatedEEPROM) after you change your settings.
In the recent snapshots there were 2 breakable changes in the eeprom layout due to #10 (-> [commit](https://github.com/ohAnd/dtuGateway/commit/4676967fa69c6d797395c5358185e681534a731f)) and #11 .

Means e.g. from 1.5.0028 to 1.5.0029 or from 1.4.0004 to 1.5.0029 the settings are lost and you have to reconfigure. With further updates there no reconfigure needed, if there is no new feature that needs a change in the user settings layout.

If you experience something other, maybe you can digging deeper and make some repetitive tests with the same firmware. And let me know the result.

thx

LukeTheMast3r, 2024-05-18:
Hi Andreas,

thanks for that information. For me thats okay. -so far :)

One more question:
Would it be very difficult to like save / backup all Fields (without wifi) to a textfile then download it to the PC.
And load it after setting up the gateway with Wifi Channel and password? 

Don't know which features will be added in the future.
If there are more and more I think having to enter all the parameters again and again will eventually become very time-consuming.

Have a nice evening
Lukas

ohAnd, 2024-05-18:
Hi Lukas,

yes, it should be a feature, should the project more growing ;-)

There are useful solutions to implement things like that.
I could be something for a 2.x version, what also needs a bigger refactoring, to have a better maintainability.
At this time, I think, the effort for the user is practicable. (on the release channel the frequency of breakable changes is very low)

Based on this ... feel free to create a dedicated feature request separately, then we can hold it in the backlog ;-)

This issue here seems be solved, therefore I will close it.

closed 2024-05-18 by ohAnd

LukeTheMast3r, 2024-05-19:
Hi Andreas,

I can only open up new issues. There is no second selection for "Feature Request" like on other Git Projects here.

Or is it enough to open up a new issue with Title: "Feature request ....."
And as input the git feature request template

Sorry i am relatively new to git so excuse my unqualified answers  ....

Greetings
Lukas

ohAnd, 2024-05-19:
Hi Lukas,

due to the kind of steering stuff like that in github, it will be more easier with a later release in main. (I had not configured with the start of this project in github)

Meanwhile you can use [this](https://github.com/ohAnd/dtuGateway/blob/ohAnd-patch-1/.github/ISSUE_TEMPLATE/feature_request.md) as an orientation.

(means simply raise a new issue with a useful explanation and this can then be tagged with "feature")

Thx