ohAnd/dtuGateway #16: Question: what is "dtu host IP in your local network:"?

hirbelo, 2024-05-18:
Hello,
 very nice project. 
I would also like to read out the values ​​of the inverter. 

But what should be entered in “dtu host IP in your local network:”? 

The inverter has its own WiFi and doesn't know my local network.
Gruß Oli

ohAnd, 2024-05-18:
Hi @hirbelo ,

that is the IP of the Hoymiles Inverter inside your local wifi - should be discoverable over your router.

The inverter has to wifi connections one for the own AccessPoint that you can reach with the hoymilesInstaller app and the other one connects with your wifi to reach the hoymiles servers (if you want). Should be done in the installation process of your inverter.
The second one gets the needed IP from your router.

Gruss
Andreas

hirbelo, 2024-05-18:
OK, thanks @ohAnd  for the quick reply.
I don't have an installation.
I'm currently looking at the values ​​via the S-Miles app without a user account - so only via inverter's own access point.
Do I understand correctly - I can't use the project without an S-Miles account?

ohAnd, 2024-05-18:
Hi @hirbelo ,

no, there is usually no S-Miles account directly needed.
The usual installation is something like this:

inverter <-> local wifi <-> internet <-> hoymiles server
                      local wifi <-> dtu gateway
in parallel it is possible over the served Accesspoint of the inverter
inverter <-> your smartphone/ tablet (S-Miles Installer App)

Therefore the dtugateway connects directly with the inverter, only in the internal network and without any external connection.

"But" in my experiences the first installation needs this account to connect the inverter with your local wifi... to make it available in the local network and also to get firmware updates by hoymiles.

Hope it answers your question.

closed 2024-06-13 by ohAnd