ohAnd/dtuGateway #17: automatic snapshot update does not work

LukeTheMast3r, 2024-05-18:
Hi Andreas,

so at first thank you for integrating mqtt and responding that fast.
Great :)

There is always a "but":
So here it is:

Noticed it from 1.5.00129 to 1.5.0030 today afternoon.
And now from 1.5.0030 to 1.5.0031

The automatic update which is shown in the web interface starts. The Countdown reaches 0. The ESP resarts. And the version stays the same.

Manual update works fine 

Hope that information will help to imrpve your work ;)

Have a nice Evening.

Lukas

ohAnd, 2024-05-18:
Jep, thanks for confirmation ...

I have the same issues on my side - but it is not "not" working - it's working only sometimes due to occuring of stack overflow issues, when it fails ;-)
That means after 2-3 tries it usually works.

It's needs some more and deep investigation, what the root cause is.

During that phase the manual way, is the backup:
- download file from release/ snapshot channel (or NEW: if there is a feature branch then from the actions and there the produced artficat)
- go with link below and upload the *.bin-file
[image]

ohAnd, 2024-05-22:
stabilized with _1.5.0035_develop_ 
after installing this all further updates should be more stable installable with direct OTA option

closed 2024-05-22 by ohAnd