ohAnd/dtuGateway #19: Issue? Client connection closed: timeout

LukeTheMast3r, 2024-05-19:
Good morning,

mqtt works well with the new ip:port setting.

Don't know if the next thing that's an issue:

In my IoBroker Debug infos there are lots of connections "Client [dtuGateway] connection closed: timeout"
See screenshot.

[image]

It does not affect the function. Only want to know if thats normal.
No other device does that. Like my shellies & Tasmota and so on.

Hope you can improve your work with that information :)

> excerpt from the serial terminal as information:

```

+++ update at remote: 19.05.2024 - 10:19:54 - uptodate: 1 --- wifi rssi: 64 % (DTU->Cloud) - 100 % (Client->AP) 

power limit (set): 40 % (101 %) --- inverter temp: 30.10 °C 

 	 |	 current  |	 voltage  |	 power    |        daily      |     total     |

grid	 |	   1.38 A |	 235.20 V |	 326.30 W |	    0.810 kWh |	  292.628 kWh |

pv0	 |	   5.19 A |	  33.20 V |	 172.30 W |	    0.433 kWh |	  148.361 kWh |

pv1	 |	   5.16 A |	  33.40 V |	 172.30 W |	    0.377 kWh |	  144.267 kWh |

>>>>> 05s task - state --> local: 19.05.2024 - 10:19:58 --- NTP: 10:19:46 --- RSSI to AP: 'B2': 100 %

>>>>> 05s task - state --> local: 19.05.2024 - 10:20:03 --- NTP: 10:19:51 --- RSSI to AP: 'B2': 100 %

>>>>> 05s task - state --> local: 19.05.2024 - 10:20:08 --- NTP: 10:19:56 --- RSSI to AP: 'B2': 100 %

>>>>> 05s task - state --> local: 19.05.2024 - 10:20:13 --- NTP: 10:20:01 --- RSSI to AP: 'B2': 100 %

>>>>> 05s task - state --> local: 19.05.2024 - 10:20:18 --- NTP: 10:20:06 --- RSSI to AP: 'B2': 100 %

scan for wifi networks done: 5 wifi's found

>>>>> 05s task - state --> local: 19.05.2024 - 10:20:23 --- NTP: 10:20:11 --- RSSI to AP: 'B2': 100 %

>>>>> 31s task - state --> local: 19.05.2024 - 10:20:24 --- NTP: 10:20:12

realData  - got remote (1716110425):	19.05.2024 - 10:20:25

GetConfig - got remote (1716110425):	19.05.2024 - 10:20:25

GridV check - 1 : 238.10 V - with: 238.10

GridV check - 2 : 238.80 V - with: 238.10

GridV check result: 0

Attempting MQTT connection... connected

sent values to mqtt broker

+++ update at remote: 19.05.2024 - 10:20:25 - uptodate: 1 --- wifi rssi: 64 % (DTU->Cloud) - 100 % (Client->AP) 

power limit (set): 40 % (101 %) --- inverter temp: 30.30 °C 

 	 |	 current  |	 voltage  |	 power    |        daily      |     total     |

grid	 |	   1.37 A |	 237.10 V |	 326.30 W |	    0.812 kWh |	  292.630 kWh |

pv0	 |	   5.21 A |	  33.10 V |	 172.30 W |	    0.434 kWh |	  148.362 kWh |

pv1	 |	   5.18 A |	  33.30 V |	 172.30 W |	    0.378 kWh |	  144.268 kWh |

>>>>> 05s task - state --> local: 19.05.2024 - 10:20:28 --- NTP: 10:20:16 --- RSSI to AP: 'B2': 100 %

>>>>> 05s task - state --> local: 19.05.2024 - 10:20:33 --- NTP: 10:20:21 --- RSSI to AP: 'B2': 100 %

>>>>> 05s task - state --> local: 19.05.2024 - 10:20:38 --- NTP: 10:20:26 --- RSSI to AP: 'B2': 100 %

>>>>> 05s task - state --> local: 19.05.2024 - 10:20:43 --- NTP: 10:20:31 --- RSSI to AP: 'B2': 100 %

>>>>> 05s task - state --> local: 19.05.2024 - 10:20:48 --- NTP: 10:20:36 --- RSSI to AP: 'B2': 100 %

>>>>> 05s task - state --> local: 19.05.2024 - 10:20:53 --- NTP: 10:20:41 --- RSSI to AP: 'B2': 100 %

>>>>> 31s task - state --> local: 19.05.2024 - 10:20:55 --- NTP: 10:20:43

realData  - got remote (1716110456):	19.05.2024 - 10:20:56

GetConfig - got remote (1716110456):	19.05.2024 - 10:20:56

GridV check - 1 : 238.10 V - with: 238.10

GridV check - 2 : 238.80 V - with: 238.10

GridV check result: 0

Attempting MQTT connection... connected

sent values to mqtt broker

+++ update at remote: 19.05.2024 - 10:20:56 - uptodate: 1 --- wifi rssi: 64 % (DTU->Cloud) - 100 % (Client->AP) 

power limit (set): 40 % (101 %) --- inverter temp: 30.40 °C 

 	 |	 current  |	 voltage  |	 power    |        daily      |     total     |

grid	 |	   1.38 A |	 235.30 V |	 326.30 W |	    0.816 kWh |	  292.634 kWh |

pv0	 |	   5.21 A |	  33.10 V |	 172.30 W |	    0.436 kWh |	  148.364 kWh |

pv1	 |	   5.18 A |	  33.30 V |	 172.30 W |	    0.380 kWh |	  144.270 kWh |

>>>>> 05s task - state --> local: 19.05.2024 - 10:20:58 --- NTP: 10:20:46 --- RSSI to AP: 'B2': 100 %

>>>>> 05s task - state --> local: 19.05.2024 - 10:21:03 --- NTP: 10:20:51 --- RSSI to AP: 'B2': 100 %

>>>>> 05s task - state --> local: 19.05.2024 - 10:21:08 --- NTP: 10:20:56 --- RSSI to AP: 'B2': 100 %

>>>>> 05s task - state --> local: 19.05.2024 - 10:21:13 --- NTP: 10:21:02 --- RSSI to AP: 'B2': 100 %

start compare: 1.5.0031 - 1.4.0004

>>>>> 05s task - state --> local: 19.05.2024 - 10:21:18 --- NTP: 10:21:07 --- RSSI to AP: 'B2': 100 %

scan for wifi networks done: 5 wifi's found

>>>>> 05s task - state --> local: 19.05.2024 - 10:21:23 --- NTP: 10:21:12 --- RSSI to AP: 'B2': 100 %

>>>>> 31s task - state --> local: 19.05.2024 - 10:21:26 --- NTP: 10:21:15

realData  - got remote (1716110487):	19.05.2024 - 10:21:27

GetConfig - got remote (1716110487):	19.05.2024 - 10:21:27

GridV check - 1 : 238.10 V - with: 238.10

GridV check - 2 : 238.80 V - with: 238.10

GridV check result: 0

Attempting MQTT connection... connected

sent values to mqtt broker

```

Lassa333, 2024-05-19:
Hi,
The FW 1.5.0031 works great!! With this version it is no problem to set a a MQTT-connection and gather the values:

[image]

With the imported values from the HM-inverter into homeassistant, one is now free to do what ever with the data (visualisation, maths, statistics,...)

Thx so much for this step forward :)

ohAnd, 2024-05-19:
@LukeTheMast3r please try the last [snapshot](https://github.com/ohAnd/dtuGateway/releases/tag/snapshot) - the mqtt connection to your broker should be staying alive now

if it's ok, then please close this issue

@Lassa333 thanks for the feedback

LukeTheMast3r, 2024-05-20:
Thanks for your great work :)

closed 2024-05-20 by LukeTheMast3r