ohAnd/dtuGateway #23: FeatureRequest: OLED connection to show crucial values of the inverter

ohAnd, 2024-05-25:
Hi Andreas,

it would be great to add an OLED-Display to show crucial values out of the inverter. With this your gateway is totally suitable for the new Hoymiles-Flagship. And a perfect replacement for OpenDTU/AhoyDTU

I could provide a picture of the values a ahoyDTU shows on the display

BR

_Originally posted by @Lassa333 in https://github.com/ohAnd/dtuGateway/issues/11#issuecomment-2128832387_

Lassa333, 2024-05-26:
Hi Andreas,

thanks for opening a new feature request. 

It would be great to offer the possibility for all guys using already a DTU to use it further. Also when someone changes from another inverter to an HMS-xxxW-2t, there would be no need to buy new hardware. So I would like to propose the following:

*	Microcontroller: ESP8266 (n.b.: maybe a ESP32 would be more future-proof?)
*	Display:  SH1106 (OLED, 1,3”, I2C, 128x64 pixels, e.g. https://www.az-delivery.de/en/products/1-3zoll-i2c-oled-display)

For the Microcontroller-display connection I would propose the use PIN D1 (SCL / Clock) and D2 (SDA / data). 

With the setup everybody can change over from a working openDTU / AhoyDTU to a wifiDTU. Or start from scratch with hardware to cover all HM inverters. Also there are already pre-configured kits on the market, which could be than also used for the wifiDTU (e.g. https://shop.blinkyparts.com/en/detail/473f89a23c3749529a2a1be80f3bc1bf )

BR

ohAnd, 2024-06-01:
@all

the mentioned [OLED display](https://www.az-delivery.de/en/products/1-3zoll-i2c-oled-display) ist now integrated as a first try... see also issue attached feature branch
(currently based on the dev branch, means only for ESP8266 - therefore ESP32 version later, after the full migration)

[image]

For all who want to give a try -> download the bin file at https://github.com/ohAnd/dtuGateway/actions/runs/9330043726 from the artifact section and update manually.

screen content and addtional features are described in the [readme](https://github.com/ohAnd/dtuGateway/tree/23-featurerequest-oled-connection-to-show-crucial-values-of-the-inverter?tab=readme-ov-file#regarding-dtu)

Lassa333, 2024-06-01:
Hi @ all,

 

wow, great job, Andreas! 🎯

 

At the moment I am running the newest firmware 1.9.0009_migrate_ESP32 on my ESP8266 – and I can confirm that this branch works fine. I will also check for the mqtt connection. Just to proof this functionality

 

I am so eager to test this Version also on an ESP32: I rebuilt my hardware already, so I will have to wait a bit longer

 

BR

ohAnd, 2024-06-01:
base integration done and merged with develop

closed 2024-06-01 by ohAnd

Lassa333, 2024-06-01:
Nice!

MQTT works stable. Integration in Homeassistant is successfully done 😊