ohAnd/dtuGateway #26: Verbindung zu Home Assistant über MQTT nicht möglich

denkandich, 2024-06-02:
Hallo,
habe einen Home Assistant 12.3 auf einem Synology in einer VM laufen. Habe einen HMS-800W-2T auf einem 286 installiert und ich bekomme die Werte von meinem Hoymiles. Wollte jetzt über MQTT die Daten zu meinem Home Assitant schicken, aber das funktioniert nicht. Mein Tasmote Zähler funktioniert, und mein HMS-1600 bringt auch seine Daten an den MQTT. Habe erst die Version 1.5.0006 versucht und dann den letzten Snapshot 1.5.0037. Bei beiden bekomme ich keine Daten, richtiger User und PW. Habe mir jetzt den MQTT Explorer installiert und hier sehe ich keinerlei Daten die vom dem HMS-800 kommen würden.
Also was mache ich falsch, eigentlich sollte die xxx37 da ja sauber funktioenren.

ohAnd, 2024-06-02:
Hi @denkandich ,

MQTT läuft bei mir in meiner "einfachen" Testumgebung ohne HA... soweit auch stabil. 
(btw. aktuell sind snapshot und release inhaltlich gleich) 

Kannst du bitte prüfen, ob deine IP und Port zu deinem Broker korrekt eingetragen sind und damit auch gespeichert wurden. Ggfs. einfach die Webseite neu laden, damit die tatsächlich gespeicherten Daten im settings Bereich zu sehen sind.

Weitere Analyse... ESP an USB an Rechner und mit Serial Monitor die Outputs anschauen. Hier sollte nach dem Reboot die Meldung Mqtt connected sichtbar werden, bzw. wiederholend der Versuch zu verbinden, wenn die Verbindung noch nicht steht...

BR

denkandich, 2024-06-02:
Hallo,habe jetzt festgestellt, über den MQTT Explorer kommen Meldungen under dem Label HMS800. Aber der Home Assistant ignoriert dies, ich kann es nicht einrichten. Den HMS1600 und den Tasmoto Zähler hat er gefunden und auch eingerichtet. Im MQTT sagt er nur 2 Geräte und der HMS800 wird nicht gefunden. Kann man den irgendwie noch einrichten, Meldungen kommen vom HSM800.

ohAnd, 2024-06-02:
Hi! 

Was meinst du mit 'HMS800'? Ist das die main topic anstelle von 'dtu1' (default), die du in den Einstellungen von dtuGateway angegeben hast?

Kann dir hier leider nicht folgen... Vielleicht kannst du einen Screenshot von deinen Einstellungen bzgl Mqtt Binding posten... Also IP Port main topic... 

Mir ist aus deiner Info noch nicht ganz klar, ob tatsächlich Daten vom dtuGateway bei deinem Broker ankommen. 

Gruss

denkandich, 2024-06-02:
HMS800 habe ich in die Main Topic eingetragen, im MQTT Explorer sehe ich die MQTT Meldung unter dem Reiter HMS800. Einstellungen habe ich die IP-Adresse von meinem NAS wo die virtuelle Maschine der HomeAssistant läuft. 192.168.178.10:1883
User ist der mqtt-user und das eingerichtete PW im MQTT. und der main Topic habe ich schon vieles getestet.

ohAnd, 2024-06-02:
Dann scheint ja die Verbindung vom dtuGateway zu deinem Broker i.O. zu stehen ...
HA seitig kann ich da leider nicht helfen, aber vielleicht wissen ja @Lassa333 oder @smarthomehomebase hier weiter ?
(ich denke, hier muss es neben der AutoDiscovery -> vgl. #22 , auch eine manuelle Config geben, um dem HA für das entsprechende item (hier der Inverter) die mqtt topic und deren Inhalt zum mapping mitzugeben)

Lassa333, 2024-06-02:
@ denkandich: 

Ich fasse nochmal kurz deine Situation zusammen:

*	Dein HMS-800W-2T Inverter sendet die Daten in dein Wlan
*	Du hast einen ESP8266 mit der FW 1.5.0037 stabil in deinem Wlan am Laufen
*	In der FW ist MQTT aktiviert, Dein MQTT-Topic hast du „HMS800“ benannt, User und Passwort sind gesetzt
*	In deiner HA Instanz empfängt der MQTT-Explorer auch die Inverter-Daten im Topic „HMS800“  

Dein Problem: Du „siehst“ die Inverter-Daten, kannst sie aber weder speichern noch nutzen: hab ich das richtig zusammengefasst?

Wenn dem so ist, füge deiner „configuration.yaml“ folgendes hinzu:

Anmerkung: Formatierung bitte genauso übernehmen + „dtu1“ musst du mit „HMS800“ ersetzten 

Das erzeugt 3 neue „Sensoren“ (Gesamtertrag, Tages-Ertrag, aktueller Ertrag des Grids), deren Werte HA für 10 Tage speichert. Und du kannst die Werte auswerten, anzeigen, ins Dashboard einbinden usw. – Eben dass, was man halt so haben will 😉

Ich hoffe, das löst dein Problem

BR

Lassa333

denkandich, 2024-06-03:
Hallo, danke für die Rückmeldung, irgendwie fehlen in deiner Nachricht die Einträge, kann das sein ?.

denkandich, 2024-06-03:
Irgendwie sehe ich den Anhang nicht, mache ich was falsch, bin angemeldet.

denkandich, 2024-06-03:
[image]
Habe es eingetragen, aber es funktioniert nicht. Keine Ahnung warum

denkandich, 2024-06-03:
Die Topic habe ich natürlich auf dtu1 gestellt.

denkandich, 2024-06-03:
Danke für die Geduld, habe es formatiert, aber der Sensor wird nicht angezeigt. Hier mal was im MQTT Explorer kommt.
[image]
Keine Ahnung was ich noch machen soll. Der HMS1600 und mein Tasmota für meinen Stromzähler wurden sofort erkannt. Langsam verliere ich die Geduld. Dir eine sehr großes Danke.

Lassa333, 2024-06-03:
Ohje! So aus der Ferne ist das echt nicht so leicht 

Vielleicht als letzten Versuch: Kopier dir mal den Inhalt der Beispiel yaml von dieser Seite ( https://www.home-assistant.io/integrations/sensor.mqtt/ ) in deine yaml. Tausche dann einfach "home/bedroom/temperature" gegen "dtu1/grid/totalEnergy" aus und versuchs nochmal.

smarthomehomebase, 2024-06-04:
Hat bei mir auch erst nicht funktioniert, dann habe ich in den bindungs bei mqtt main topic for this dtu den / entfernt. Das hat zum Erfolg geführt.
[image]
[image]

smarthomehomebase, 2024-06-04:
Gibt es den Wirkungsgrad/Einstrahlung auch als Sensorwert?

denkandich, 2024-06-08:
Hallo, habe jetz auf den letzten Snapshot gewechselt, aber das Interface schaut noch alt aus. In der Beschreibung lese ich
New 'dtuGateway web config interface' to configure all possible config parameter in one shot. Wo kann ich das finden ?.

Lassa333, 2024-06-09:
Hi denkandich,

versuch doch bitte mal den Snapshot von gestern abend. Andreas war tätig und es könnte sein, dass dieser dein Probleme löst 😉

Ich würde mich freuen, über deine Fortschritte Bericht zu bekommen - Andreas sicherlich auch

denkandich, 2024-06-09:
Wie schon im anderen Beitrag, wie komme ich zu dem Snapshot, bin wenig in der Sache drin, finde nur den alten Snapshot.

Lassa333, 2024-06-09:
https://github.com/ohAnd/dtuGateway/actions/runs/9407361930
-------------------------------------------------------------------------------------------------
FreeMail powered by mail.de - MEHR SICHERHEIT, SERIOSITÄT UND KOMFORT

denkandich, 2024-06-09:
Das ist das Problem, ich kann mit dem Link nichts machen, da ist keine bin die ich laden kann. Scheint ich mache das zu wenig.

ohAnd, 2024-06-09:
Hi @denkandich ,

"einfach" das artefact herunterladen entpacken -> da ist das bin-file enthalten. (werde schauen den Namen von "artefact" auf sinnvolles noch zu ändern)

[image]

Da da das Thema noch recht "frisch" ist wird es erstmal auf einem feature branch bearbeitet und ist damit noch nicht ein latest snapshot, bis es eben grundsätzlich rund erscheint... Daher auch nur über diesen Weg zu erreichen. So ist es möglich das z.B. @Lassa333 so früh als möglich ins Testing (und spezifieren nochmal danke dafür ;-) ) gehen kann und trotzdem nicht selbst kompilieren braucht.

Gruss

btw. wenn es geht, bitte Fragen, Rückfragen, Anregungen im passenden Issue zu halten ggfs. ein neues eröffnen... wird sonst schnell schwierig dem Inhalt zu folgen

Lassa333, 2024-06-09:
Guter Hinweis, ich habe gerade auch im anderen Post geantwortet. Ich stells hier nochmal ein - und wir testen hier weiter, OK?

Du hast die Version 1.6.0019_22 aktiv?

Bei der Version kannst du wieder (wie gewohnt) über die Einstellungen arbeiten. Versuch doch bitte mal folgendes:

in der Firmware: Unter Einstellungen -> bindings ganz runter scrollen und den Haken bei "Homeassistant Auto discovery" setzten. Speichern.

Bekommst du nun deinem Homeassistant eine neue Benachrichtigung, dass ein neues Gerät entdeckt wurde?

ohAnd, 2024-06-13:
due to https://github.com/ohAnd/dtuGateway/issues/22#issuecomment-2156687287 - done

closed 2024-06-13 by ohAnd