ohAnd/dtuGateway #27: Uhrzeit im Oled ist falsch ( 1 Stunde zurück )

checkersky, 2024-06-03:
Moin, ich bentze die Version 1.5.0006 mit einem Oled Screen dabei ist mit aufgefallen das im Oled Screen
die Uhr nicht richtig angezeigt wird.

Im Oled Screen zeigt die Uhr immer eine Stunde weniger an als es wirklich ist.

Im web interface ist die Uhrzeit aber richtig !

Also Webinterface z.B. 11:30 Oled zeigt aber 10:30 an

ohAnd, 2024-06-04:
Hi @checkersky ,

jep, danke für die Info. Aktuell ist nur die fixe timezone (CET) enthalten.

=> Automatische Umschaltung Sommerzeit, sowie user-bezogene Einstellung der TImeZone lasse ich hier mal als FeatureRequest stehen...

btw. im Webinterface weiss es der browser ;-), da hier von GMT kommend per javascript umgerechnet wird.
Vorübergehend würde ich fix auf CEST für die nächsten Devs/ Releases umstellen.

BR

ohAnd, 2024-06-05:
Hi @checkersky ,

mit dem [latest Snapshot](https://github.com/ohAnd/dtuGateway/releases/tag/snapshot) ist über das neue ConfigWebInterface die Zeitdifferenz direkt user-spezifisch editierbar. Siehe auch [readme](https://github.com/ohAnd/dtuGateway/tree/develop?tab=readme-ov-file#regarding-base-framework)

Gerne testen.

BR

checkersky, 2024-06-06:
Funktioniert ! Danke

closed 2024-06-06 by ohAnd