ohAnd/dtuGateway #29: IP/port to MQTT Platform (https://www.hivemq.com)

smarthomehomebase, 2024-06-04:
Für die Anbindung an die MQTT Platform hivemq.com wir die Url 12345678.s1.eu.hivemq.cloud und der Port 8883 benötigt. Gibt es eine Möglichkeit das im dem Feld einzugeben?

ohAnd, 2024-06-04:
Hi @smarthomehomebase ,

derzeit gibt es auch hier noch die Längenbeschränkung in der Eingabe und EEPROM-Speicher. (also nur IP eingebbar).

Aktuell baue ich die userConfig um (file basiert statt EEPROM), so dass dann auch längere Domainnamen möglich sein sollten.
Damit sollte dann auch eine Domain anstelle der direkten IP erreichbar werden ...
Notation dann "12345678.s1.eu.hivemq.cloud:8883"

BR

ohAnd, 2024-06-05:
Hi @smarthomehomebase ,

mit [latest Snapshot](https://github.com/ohAnd/dtuGateway/releases/tag/snapshot) ->   [dtuGateway_snapshot_1.6.0038_develop.bin](https://github.com/ohAnd/dtuGateway/releases/download/snapshot/dtuGateway_snapshot_1.6.0038_develop.bin) ist das Layout umgestellt und auch die Länge für IP/ Domain auf 128 Zeichen erhöht.
Daher gerne mal testen ... und kurzes Feedback.

BR

smarthomehomebase, 2024-06-06:
Hi Andreas,
die Domain und den Port kann ich eintragen. Allerdings wird von der hivemq cloud noch TLS benötigt. Bei meiner anderen OpenDTU von tbnobody werden mit TLS die Daten übertragen.

ohAnd, 2024-06-19:
Hi @smarthomehomebase ,

nur als Zwischenstand - Thema ist noch auf der Liste - nach ein paar Tests braucht es ein grösseres Refactoring, da das TLS ein wenig mehr Speicher benötigt, als mit dem aktuellen Aufbau verfügbar ist.

D.h. z.B. ohne aktiven Webserver ist es problemlos möglich die Verbindung zu hivemq aufzubauen und entsprechend messsages am Broker abzusetzen. Daher nun erst Refactoring inkl. Transfer zu ESP32 ...

BR

PR #48 (2024-09-03): implements IP/port to MQTT Platform (https://www.hivemq.com) #29 - fo…
…r ESP32 only

closes #29
closed 2024-09-03 by ohAnd