ohAnd/dtuGateway #31: Display GAGC9A01 funktioniert nicht

checkersky, 2024-06-14:
Ich habe die Version 1.6.0038_develop installiert und die vorher mit einem Oled benutzt was auch super funktioniert hat.
Jetzt habe ich ein GAGC9A01 display gekauft bekomme es aber nicht zum laufen.

In der Config habe ich das Display auf 1 gestellt 

Ich benutze einen NodeMCU V3 und habe den wie folgt angeschlossen:

Pin Display:    Pin ESP:

VCC,           =   3V
GND           =  Gnd
SCL             =  SK
SDA            =  S1
DC              =  D3
CS               =  CS
RST             = 3V
BLK             = nicht angeschlossen

Wenn ich das so anschließe fährt der ESP gar nicht mehr hoch.

Habe ich beim anschließen einen Denkfehler gemacht oder ist das Display defekt ?

ohAnd, 2024-06-14:
Hi @checkersky ,

deine Belegung, wenn NodeMCU V3, scheint für mich korrekt. Meine Display-Variante hat keine Ausführung für BLK (back light control) daher wäre das noch ein Punkt - nachdem was ich dazu aber gesehen habe, kann dieser Pin auch offen bleiben, da dann volle Helligkeit.

Ich würde fast auf einen Defekt schliessen... Habe 3 Stück davon und mit den verschiedenen Boards getestet, alles i.O.. 

Da ich noch eine D1 NodeMCU die ich noch nicht getestet habe und eine andere Belegung hat, vielleicht kannst du folgende Belegung noch testen...

Pin Display: Pin ESP:
VCC, = 3,3V
GND = Gnd
SCL =       -> GPIO14  => D13
SDA =       -> GPIO13  => D11
DC =        -> GPIO0    => D8
CS =        -> GPIO15  => D15
RST = 3,3V
BLK = nicht angeschlossen

Wäre zumindest einen Versuch wert ;-)

BR

checkersky, 2024-06-14:
Danke, dann ich bestelle ein neues Display.

checkersky, 2024-06-20:
Hallo, neues Display ist da geht leider auch nicht… 
Ich habe mein Oled wieder eingebaut aber auch da gibt es ein kleines Problem das Display ist nicht richtig zentriert. Auch Geht der Screensaver nicht das Display bleibt immer immer an der selben Stelle.

Soll ich dafür lieber einen neuen Bug Report erstellen?

[image]

checkersky, 2024-06-20:
Upps, versehentlich geschlossen

ohAnd, 2024-06-20:
Hi @checkersky ,

sorry, mit der aktuellen Beschreibung des Problems lässt sich aus der Ferne noch nicht viel sagen, sondern nur vermuten...
Wie schon gesagt funktionieren bei mir beide Display-Typen (dev snapshot), über die in der Readme gezeigten Boards und PIN-Belegungen.

Kannst Du bitte mal die genaue Bezeichnung deines Boards posten, welches du verwendest und welche Pin-Belegung(en) du hier für die beiden Displays versucht hast? ggfs. auch ein Foto vom Board/ "Verdrahtung"?

Zum Versatz auf Deinem Display:
Kann ich mir spontan nicht erklären, auch wenn du hier eine andere Variante (blaues OLED und Pin-Belegung GND/VCC getauscht) einsetzt, muss der Treiber SSH1106 die Pixel gleich umsetzen...

Hier vielleicht die Bitte an andere OLED User von dtuGateway "kurze Info", ob dieses Problem ebenso schon gesehen wurde ...

(lt. deinem Foto hängt das Display "merkwürdig schräg" auf der Platine... mal geschaut, wie die Lötstellen auf der Rückseite für den Displayanschluss aussehen? wenn hier ein Fehler vorliegt könnte es vielleicht die Verschiebung erklären ... *ratenderweise*)

Der Screensaver läuft in der Art das alle 60 sec die gesamte Anzeige um 1 pixel verschoben wird rechts, runter, links, hoch... das sollte auch mit dem Fehler sichtbar, da ja scheinbar die eigentliche Anzeige sichtbar wird...

BR

checkersky, 2024-06-23:
Hi,

ich habe mal ein paar Bilder abgehangen von den Displays die ich hier habe.
Von der Oleds habe ich ein blaues und ein weißes beide haben den Versatz. Getestet sind die Displays mit einem anderen Projekt an eine Pi da funktionieren die beiden ohne Probleme.

Zu den GAGC9A01 : Ageschlossen habe ich beide nach dem Plan den ich im ersten Post gepostet habe.
Bilder von Anschluss habe ich nicht aber ich habe Bilder von dem Esp und der Belegung abgehangen.

checkersky, 2024-06-23:
Ich habe gerade das Neuste Beta für den ESP 32 auf meinen ESP32 Mini geflasht und auch da ist das Display verschoben.
Hast du ein Pinout für das GAGC9A01 für den ESP32 ? dann würde ich das nochmal testen eventuel geht es ja mit dem ESP32

ohAnd, 2024-06-24:
Hi @checkersky ,

danke schonmal für das hartnäckig bleiben ...
Konnte das Verhalten beim TFT mit verschiedenen builds (remote vs. lokal) ebenso beobachten... Muss hier aber noch weiter analysieren.
OLED Problematik hatte ich so noch nicht - aber vielleicht zeigt sich bei der weiteren Analyse auch hier noch was ;-)

BR

checkersky, 2024-06-24:
Hi, kein Problem mache ich gerne da ich dein Projekt echt genial finde =)

Hast du denn schon ein Pinout für den ESP32 und dem TFT ? Das würde ich noch gerne ausprobieren.

ohAnd, 2024-06-25:
Hi @checkersky,

die findest du hier [https://github.com/ohAnd/dtuGateway/tree/migrate_ESP32?tab=readme-ov-file#installation--update](https://github.com/ohAnd/dtuGateway/tree/migrate_ESP32?tab=readme-ov-file#installation--update)

Aaaaber 😉 da steckt aktuell das (oder auch ein) Problem... Ich bin grad am Refactoring dran... Und ein kurzer Test lässt bei mir das TFT auch nur "glimmen"... Bin aber grad an anderer Stelle dran, daher Geduld 😀

Was aber auf jeden Fall für beide Varianten (8266/32) geht ist das OLED, zumindest in meinen Konfigurationen ohne Verschiebung. 

Soweit als Zwischenstand. 

BR

ohAnd, 2024-07-06:
hi @checkersky ,

sorry hat einen Moment gedauert (aufgrund des Refactorings erstmal hinten angestellt) ... konnte den Fehler finden. (war "nur" ein Problem mit build-flags für das TFT Display, so dass auf gitHub "falsch" gebaut wurde und daher die angebotene Firmware eben das TFT nicht unterstützt hat.).

Konnte es daher aber auch "einfach" für das snapshot-release nachziehen.
Siehe -> [https://github.com/ohAnd/dtuGateway/releases/tag/snapshot](https://github.com/ohAnd/dtuGateway/releases/tag/snapshot)

Beschreibung der PIN-Belegung lt. [readme](https://github.com/ohAnd/dtuGateway/tree/develop?tab=readme-ov-file#hardware) gilt weiterhin.

Daher gerne nochmal den Versuch unternehmen.

Sorry, das schlussendlich das das Problem war.

btw. Für den ESP32/ESP8266 preDev Pfad ist das Thema dann ebenso gefixt.

BR

checkersky, 2024-07-06:
Hi, das Display funktioniert nun am ESP8266!

Leider kann ich das Gateway mit dem neuen Snapshot nicht mehr einrichten. 
Ich kann den first start nicht mehr im Browser öffnen ich kann somit das Gateway nicht einrichten.
Mache ich ein alten Snapshot drauf geht der first start wieder.

ohAnd, 2024-07-06:
Jep ;-), auch das war ein Grund für das Aufräumen...

Fixed! (snapshot path)

checkersky, 2024-07-06:
Funktioniert alles vielen dank =))

closed 2024-07-06 by checkersky