ohAnd/dtuGateway #37: Cant set Limit over MQTT

haesslo, 2024-07-22:
I have the whole thing setup and can read data over webinterface MQTT and Openhab RestAPI
However setting the Limit works with Webinterface, OpenHab RestAPI, but not when i publish a raw number to:
 dtu_8775546/inverter/PowerLimit_Set (using MQTT Expolorer)

fw version: 1.6.0007

Any thoughts ?

Thx

gg

ohAnd, 2024-07-24:
hi @haesslo ,
you're right ... it's fixed in the dev with a new snapshot.
Please check and close, if it is ok.
BR

haesslo, 2024-07-24:
Hi ohAnd

I installed last snap shot....maybe something went wrong, but now i can only set via OpenHab RestApi, but not anymore via webinterface and not via MQTT....

ohAnd, 2024-07-25:
Hi @haesslo ,

jep, that is right... web-interface and mqtt are event driven, so there is no direct concurrency - means similiar to FIFO. But openhab part is currently pulled every second.
-> if openhab is active - openhab will be the master
Is there a use case to have different sources for the power limit setting at the same time?

Nevertheless:
I will check to implement a small observer to take over only the changes from openhab to make it also event driven - maybe then only in the 2.x path (ESP8266/ESP32)

BR

haesslo, 2024-07-25:
Hi ohAnd
I dont have a usecase fpr different sources, just thought it might help on finding the problem to know what orks and what does not work.
My prefered solution would still be MQTT only, but no success so far in getting it working.....

ohAnd, 2024-07-25:
Hi @haesslo ,

I've reviewed the mqtt set path in this dev path ... and fixed it, hopefully now ;-)
And also made the openhab also event driven.
But everything only rough tested.
Maybe you can test it for all sources -> see snapshot.

Thx

haesslo, 2024-07-25:
Hi @ohAnd 

Working now !

Cheers

gg