ohAnd/dtuGateway #39: Mobile Browser show raw html code (fw version: 1.6.0008)

steffengoethling, 2024-07-26:
When i open the dtu surface on my mobile phone, i see the raw html code instead the rendered dtu gateway surface

**To Reproduce**
Steps to reproduce the behavior:
1. Go to dtu home with iPhone mobile browser (Safari, Chrome)
2. See raw html code

**Expected behavior**
Normal surface is shown

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - iPhone 14
 - OS: 17.5.1
 - Browser safari, chrome
 - Version newest

ohAnd, 2024-07-27:
Hi @steffengoethling ,

thanks for the hint ... that's weird.
I don't checked for every snapshot / release iOS based systems. (with earlier versions it definitely works)
But at least for this version, the website will be not rendered on iOS based devices (checked on iPhone/iPad) neither with safari nor with edge.
On windows with chrome/ edge/ firefox and also an a chromebook with chrome there are no issues...

Due to the upcoming change to the 2.x path of dtuGateway, maybe you want to directly change to the new path (branch: [migrate_ESP32](https://github.com/ohAnd/dtuGateway/tree/migrate_ESP32)) - that problem is not given here.

Or try some of the older releases to start.

I will check with delay what could be the problem here...

BR

steffengoethling, 2024-07-27:
Hey BR,

yes, with the older release the surface is shown on my iPhone. On my MacBook is everything normal with Chrome and the actual release. I don't want to downgrade, because i want use the new power limit functionality :-)
Thank you very much for your support and for sharing the software!

Best regards

Steffen

commit ce98d5a5 (2024-08-02): Merge pull request #40 from ohAnd/migrate_ESP32

migration to the refactored path with multi-arch and a lot of changes for
- supporting ESP8266 and ESP32
- better maintainability
- better responsive web ui
- fix #39 
- fix #28
- close #20
commit 89f6e4fe (2025-09-13): Merge pull request #51 from ohAnd/develop

releasing 2.x
- fix #49 
- fix #39 
- fix #28 
- fix #20
closed 2024-09-11 by ohAnd