ohAnd/dtuGateway #45: Bildschirmschoner bzw. lcd aus

checkersky, 2024-08-28:
Hallo,

Ist es möglich das LCD komplett auszuschalten ?
Es wäre super wenn man es konfigurieren könnte das es z.b. von 22:00 bis 6:00 uhr aus ist.

wenn das komplett ausschalten nicht geht dann vielleicht einen Screensaver wo zb nur die Uhrzeit angezeigt wird und der Rest schwarz ist.

ohAnd, 2024-08-30:
Hi @checkersky ,

das Thema habe ich auch schon auf der Liste... hab deswegen schon andere TFTs mit BLK CTRL "rumliegen" ... ;-)
Ohne backlight control wird man nur die Anzeige leer bekommen, aber das Hintergrundlicht wird aktiv bleiben.

Würde es mal versuchen in der Art auf die Liste zu nehmen:
- für alle Varianten (OLED, TFT mit/ohne BLK) Einstellung über advanced config
  - Nachtmodus aktiv/ deaktiv
  - Startzeit und Stoppzeit des Nachtmodus (erstmal nur volle Stunde)
  - screen blank oder Uhrzeit
- damit wird dann entsprechend config das display dunkel (oder eben nur leer) gesetzt oder die aktuelle Uhrzeit angezeigt

Gruss
Andreas

Frankyistda, 2024-09-08:
Kannst Du vielleicht auf die Worte "Offline" hin überprüfen, die im Diplay erscheinen? Wenn die z.B. 5x in Folge kommen, dann Anzeige aus? Ist aber alles nicht entscheidend!

ohAnd, 2024-09-10:
@checkersky 
nightmode habe ich nun im snapshot... details in der readme -> [display support](https://github.com/ohAnd/dtuGateway/tree/develop?tab=readme-ov-file#display-support)
gerne mal checken ... 

@Frankyistda 
schiebe deinen punkt mal in ein neues issue... damit dieses hier abgeschlossen werden kann...

closed 2024-09-10 by ohAnd

checkersky, 2024-09-10:
Nabend, habe es gerade ausprobiert geht alles bis auf das ausschalten des  TFT ich habe mein Pin 6 an BKL angeschlossen aber aus geht es nicht. 

ich habe folgenes TFT:

[image]

ohAnd, 2024-09-10:
=> https://github.com/ohAnd/dtuGateway/tree/develop?tab=readme-ov-file#hardware

Display Pin BLK an ESP8266 -> D6/GPIO12
oder 
Display Pin BLK an ESP32 -> D4/GPIO04

Was immer dein Pin 6 ist ;-) ?
Hängt ja vom verwendeten Board ab - wichtig ist der entsprechend der auf den Pin gemappte GPIO eben 12 (ESP8266) oder 4 (ESP32)

checkersky, 2024-09-10:
Ich meinte D6 nicht Pin 6.. da ist bei mir GPIO12.

ohAnd, 2024-09-10:
ändert sich denn die Helligkeit, wenn du die einstellungen entsprechend setzt?

mit folgendem Beispielsettings läuft es bei mir auf nem ESP8266 -> 12 und ESP32 -> 4

brightnessDay    150
brightnessNight  30
nightClock          true
nightMode         true
nightmodeStart  1320 => 22:00
nightmodeEnd    390 => 6:30

der Helligkeitsunterschied wird erst so richtig unter 100 erkennbar

checkersky, 2024-09-10:
Ich bin leider gerade nicht zu Hause aber heute morgen habe ich verschiedene Helligkeiten ausprobiert und da hat sich nichts getan.

checkersky, 2024-09-10:
Es geht !!

ich habe diese werte:

brightnessDay 150
brightnessNight 40
nightClock false
nightMode true
nightmodeStart 1320 => 22:00
nightmodeEnd 390 => 6:30

ich habe nach den übernehmen der Werte den ESP abgesteckt und kurz gewartet ( also einen Hard Reset gemacht ) dann ging es !