ohAnd/dtuGateway #47: Update in die cloud geht nicht immer

checkersky, 2024-09-03:
Hi, ich habe das Problem, dass die Updates in die Cloud nicht richtig funktionieren.

Obwohl bei mir die Cloud Pause ausgeschaltet ist, hab ich das Problem, dass wenn das Gateway läuft, ich teilweise keine Updates mehr in die Cloud bekomme.

Die letzte Person womit funktioniert hat, ist der Snapshot 1.90036 alle Versionen danach haben dieses Problem. Ich hab mal einen Screenshot angehangen, wo man das ganz gut sehen kann.

Im zweiten Screenshot sieht man auch, dass die Uhrzeit nicht mehren der Cloud Gehupdated wird .
[image]
[image]
Obwohl wir schon 14:30 Uhr haben, zeigt er noch 7:00 Uhr an.

checkersky, 2024-09-03:
[image]
[image]

checkersky, 2024-09-05:
Habe jetzt seit gestern dir Version dtuGateway_snapshot_1.9.0052_develop drauf damit funktioniert das Update in die Cloud wieder !

ohAnd, 2024-09-05:
Hi @checkersky ,

"guter" Fehler, der sich von selbst heilt... ;-)
Habe in dem Bereich schon länger nichts mehr verändert. 
War aber der Meinung das in den letzten Tagen des Öfteren im Cloud-Service etwas "komisch" war...

Generell: wenn die Werte nicht geupdated werden, hat es nach meinen Erfahrungen mit einer "Belegung" der DTU um den Zeitpunkt alle 15 Minuten + ca. 7 Minuten zu tun ... die "cloud pause" hatte ich deswegen eingebaut, und verhindert Anfragen an die DTU < 30 und +10 sec um diese Zeitpunkte (15 min Takt) ... Wenn man es versucht zu verfolgen auf der Cloudseite, dann wird sehr schnell nach den Zeitpunkten der letzte timestamp in der cloud hochgezogen, aber die dazugehörigen Werte erst ca. 7 min später dargestellt ... 
Und wenn hier dann ein Powerset und "ausserplanmäßige" Anfrage reinrauscht, dann wird scheinbar nichts zur Cloud geposted...

closed 2024-09-05 by ohAnd

checkersky, 2024-09-05:
Oh.. ich dachte das die Cloud Pause genau das gegen teil macht also KEINE Daten zur Cloud senden.

Was aber noch schön wäre wenn da nicht über das ganze Display "Cloud Pause" steht sondern so wie im kleinen Oled Display nur ein Icon an geht. Eventuell als Option in den Einstellungen ?