ohAnd/dtuGateway #49: nightmode triggered by offline dtu

ohAnd, 2024-09-10:
Kannst Du vielleicht auf die Worte "Offline" hin überprüfen, die im Diplay erscheinen? Wenn die z.B. 5x in Folge kommen, dann Anzeige aus? Ist aber alles nicht entscheidend!

_Originally posted by @Frankyistda in https://github.com/ohAnd/dtuGateway/issues/45#issuecomment-2336707017_

closed 2024-09-11 by ohAnd

Frankyistda, 2024-10-19:
Hallo Andreas @ohAnd 

Das mit den Einstellungen dazu habe ich entweder nicht richtig verstanden oder es funktioniert bei mir nicht.

Eingestellt ist:

nightMode false
nightModeOfflineTrigger true
nightmodeStart 1320
nightmodeEnd 390

Dann sollte doch zwischen 22 Uhr und 6:30 Uhr nach wiederholtem Offline das Display eigentlich ausgehen, habe ich das richtig verstanden?

Gruß
Frank

Frankyistda, 2024-10-19:
Heute habe ich das mal auf der Web Oberfläche beobachtet.

Das kam bei mir in eine Endlosschleife: try reconnect - offline - try reconnect - offline usw.

ohAnd, 2024-10-20:
> Hallo Andreas @ohAnd
>
> Das mit den Einstellungen dazu habe ich entweder nicht richtig verstanden oder es funktioniert bei mir nicht.
>
> Eingestellt ist:
>
> nightMode false nightModeOfflineTrigger true [...]

Hi @Frankyistda ,

hatte versucht das hier zu beschreiben ;-) : 
https://github.com/ohAnd/dtuGateway?tab=readme-ov-file#display-support
nightMode ist quasi der Hauptschalter - nur dann wird es überhaupt mal "dunkel" und das wann wird über die trigger definiert,
also entweder
- Uhrzeit von/bis (nightModeOfflineTrigger = false) oder
- Uhrzeit von/bis und offline (offline ist immer höherwertig als Uhrzeit) -  (nightModeOfflineTrigger = true) oder
- nur offline, wenn Uhrzeit z.B. 0:00 bis 0:00 (noch nicjht wirklich getestet ;-) ) - (nightModeOfflineTrigger = true)

Gruss
Andreas

ohAnd, 2024-10-20:
> Heute habe ich das mal auf der Web Oberfläche beobachtet.
> 
> Das kam bei mir in eine Endlosschleife: try reconnect - offline - try reconnect - offline usw.

ist hier ein wenig offtopic...

Deine Beobachtung sollte der Normalfall sein, wenn die DTU offline ist (keine Sonne/ oder angeschlossene Batterie "leer") - so das das dtuGateway regelmäßig prüft ob die DTU wieder verfügbar ist. Wie schon an anderer Stelle gesagt, ist das die Anzeige auf der WebUI um hier den Zustand der Verbindung sichtbar zu machen. Für die Offline-Aussage für den Trigger der Display-Umschaltung werden die Infos nochmal gefiltert.

commit 89f6e4fe (2025-09-13): Merge pull request #51 from ohAnd/develop

releasing 2.x
- fix #49 
- fix #39 
- fix #28 
- fix #20