ohAnd/dtuGateway #54: Im Status "try reconnect" wird die "cloud pause" nicht eingehalten (v1.9.0049)

Frankyistda, 2024-09-27:
Hallo Andreas @ohAnd !

Gestern Abend ist mir aufgefallen, dass nachdem meine Speicherbatterie keinen Strom mehr abgegeben hat, dtuGateway immer wieder versuchte sich zu reconnecten (try reconnect) und dabei die Cloud Pause nicht eingehalten hat.

Nachdem ich dann den Stecker für den esp gezogen hatte, wurden auch wieder Daten in die Hoymiles Cloud (0 W) gegeben.

Ich hatte allerdings noch die ältere Version v1.9.0049 installiert, vielleicht ist das Problem mit der aktuellen v2.0.0010 ja schon behoben.

Die habe ich heute nun geladen und werde das heute Abend noch einmal versuchen zu beobachten.

Ansonsten: Große Arbeit, tolles Teil! Kann ich nur immer wiederholen! vielen Dank für Deine Arbeit!

Gruß
Frank

ohAnd, 2024-10-02:
Hi @Frankyistda ,

danke für die Info! Bitte mal beobachten.
Mglw. wird hier nur die Darstellung bzgl. DTU status fehlerhaft überblendet. Die Pause wird intern ein wenig tiefer angesetzt.

Danke

ohAnd, 2024-10-20:
HI @Frankyistda ,

konntest Du den Punkt nochmal beobachten ? Anderenfalls gerne das Issues schliessen.

Danke

Frankyistda, 2024-10-21:
Hallo Andreas,

 

mal sehen ob man auch einfach so aus der Email heraus antworten kann.

 

Ich habe das nicht mehr beobachten können, weil ich leider das dtuGateway jetzt erst einmal außer Betrieb nehmen musste.

 

Seit ich die App von everHome benutze, funktioniert Dein dtuGateway leider nicht mehr so schön wie zuvor.

 

Es passiert immer wieder, dass DTU Offline eintritt und dtuGateway nicht mehr auf den Wechselrichter zugreifen kann. Dann connected er wieder, ist aber wieder schnell offline, gefolgt von connect usw. Am Ende wird dann ein dtu Restart durchgeführt und alles läuft eine Zeitlang wieder sauber. Bis sich das Verhalten wiederholt.

 

Außerdem bin ich mir nicht sicher, ob andersherum der Zugriff von everHome auf den Wechselrichter dadurch nicht behindert wird.

 

Leider fehlen mir da die Zusammenhänge, wie im einzelnen die ganzen Programme auf den Hoymiles zugreifen. Deshalb hatte ich mir ja den MQTT Zugriff von everHome auf Dein dtuGateway erhofft, um die dtuGateway Daten als Input für die everHome App nehmen zu können. everHome hat aber nur eine openDTU Schnittstelle dafür.

 

Da ich aber nicht schlau daraus werde, in welcher Struktur openDTU die MQTT Daten published (static und config usw.) und wie ich das in Deinem Quellcode möglicherweise abändern könnte, fällt der Workaround auch aus.

 

Vielleicht wäre in Deinem Programm statt des dauerhaften connects ein connect - Daten auslesen - disconnect - Pause - connect usw. eine Lösung, aber genaues weiß man nicht.

 

 

Gruß

Frank

 

 

 

Gesendet: Sonntag, 20. Oktober 2024 um 20:39 Uhr