ohAnd/dtuGateway #55: everHome EcoTracker MQTT Integration: Wie?

Frankyistda, 2024-10-01:
Hallo Andreas @ohAnd !

Ich entwickle mich zur Nervensäge, sorry!

Ich habe mir gerade das Produkt EcoTracker von everHome zugelegt um den Stromzähler auslesen zu können und den Ertrag des Hoymiles HMS-800W-2T dagegenzustellen.

Wenn ich jetzt den ecoTracker mit dem Wechselrichter direkt verbinde, dann funktioniert leider Dein tolles dtuGateway nicht mehr. Das hatte ich schon befürchtet.

Jetzt lässt die everHome App aber auch die Integration von OpenDTU über MQTT zu. 

Das habe ich mit dem dtuGateway auch versucht, es ist mir allerdings nicht gelungen. Vielleicht hast Du ja eine Idee für mich, was ich da falsch mache.

everHome verlangt als MQTT Konfiguration zwingend:
MQTT-Server: everhome.cloud
MQTT-Port: 1883
der MQTT-Benutzer und das zugehörige Passwort werden von everHome generiert und vorgegeben.

Jetzt habe ich in dtuGateway unter bindings eingetragen:

MQTT connection: Haken gesetzt
IP/port to MQTT broker: everhome.cloud:1883
specify user on your mqtt broker instance: 39941_5 (kommt als Vorgabe)
password for the given mqtt user: Mausefalle (kommt als Vorgabe)
MQTT main topic for this dtu: dtu_5077xxx (stand schon vorher drin)

über save und close das Menü verlassen und auch mal den Stecker gezogen um ein Reboot zu erzwingen, hat aber nichts genützt.

Es kommen leider keine Daten an.

Hast Du eine Idee?

Danke im Voraus!

Gruß
Frank

tengel07, 2024-10-01:
Hallo Frank @Frankyistda 

ich vermute mal, der Everhome MQTT-Server lässt nur die Daten durch, die von den für Everhome kompatiblen Geräten kommen.
Also im Bereich DTU aktuell nur: openDTU und AhoyDTU.
Das DTU von Andreas ist sicher noch nicht bei Everhome unterstützt - vielleicht mal bei everhome-support nachfragen.

Ich nutzte für die Auswertung einen kleinen Mini-PC (NiPoGi Mini PC), darauf läuft Ubuntu + Docker, und im Docker läuft als Container der IO-Broker, die InfluxDB und Grafana für die Auswertung. Der IO-Broker nimmt per MQTT die Daten vom Stromzähler über einen günstigen IR-Lesekopf und vom WR-DTU entgegen und speichert die Werte in der InfluxDB. Mittels Grafana sind dann passende Auswertungen erstellbar und einfach per Browser abrufbar. Auch influxdb und iobroker haben Webinterfaces zum Konfigurieren - es ist also fast nicht nötig auf dem Mini-PC sich einzuloggen.

@ohAnd : Dein DTU-Code rennt seit Mitte August diesen Jahres ohne Probleme und liefert ordentlich seine Daten ab.
Klasse Arbeit!

Viele Grüße
Torsten

ohAnd, 2024-10-02:
@tengel07/ @Frankyistda 

everHome kannte ich noch nicht, aber im Grundsatz ist es ja wie in jeder Anwendung die "fremde" Daten einbinden möchte ... die API muss bekannt sein. D.h. everHome muss "wissen" (innerhalb der SW oder wenn verfügbar durch konfigurierbare Schnittstellen) unter welcher Main-Topic sind die Daten zu finden und wie sind diese dort in Topics sortiert und "codiert" also Auflösung/ Einheit, also z.B. grid power in dtuGateway unter <main_topic>/grid/P die Inverterleistung ins Netz in kWh

Z.B. bei HomeAssistant wird das AutoDiscovery Protokoll über MQTT genutzt, was eben genau diese Konfig von aussen auf generischem Wege möglich macht - btw. auch Openhab hat für die AD eine Umsetzung, so das es in gleicher Form auch dort verwendet werden kann.

Daher vielleicht schauen, ob everHome nicht auch diese Variante zulässt.

ohAnd, 2024-10-02:
@Frankyistda 

noch dazu

> Wenn ich jetzt den ecoTracker mit dem Wechselrichter direkt verbinde, dann funktioniert leider Dein tolles dtuGateway nicht mehr. Das hatte ich schon befürchtet.
> 

Was meinst du mit ecoTracker direkt mit dem Wechselrichter verbinden? Der EcoTracker ist doch "nur" ein IR-Lesekopf für den Stromzähler... Und warum ist dadurch dann keine Verbindung vom dutGateway zum Hoymiles mehr möglich ...?

Oder meinst Du die App von everHome die dann auch über GoogleProtoBuf direkt auf den Hoymiles zugreifen kann?

Mir fehlt hier die "Verbindung" - aus allg. Interesse kannst Du dein Setup noch ein wenig erklären... ;-)

Frankyistda, 2024-10-03:
@ohAnd Hallo Andreas,

ja, das muss ich berichtigen.

Ich habe bei gleichzeitiger Anbindung des Hoymiles Wechselrichters an dtuGateway und der everHome Cloud via App (ich denke Du hast Recht, der ecoTracker hat damit nichts zu tun) festgestellt, dass dtuGateway oft "Offline" meldet.

Das passiert quasi minütlich immer zu jeder halben Minute für 1-2 Sekunden und manchmal (so alle 10 Minuten) für ca. 1,5 Minuten. Ich vermute, da ruft everHome die Daten ab.

Mittlerweile habe ich aber beides (dtuGateway und everHome) wieder gleichzeitig laufen, weil ich festgestellt habe, dass die Daten trotzdem an everHome übertragen werden. Ich hatte vorher befürchtet, dass everHome ansonsten nicht funktioniert.

Mir entzieht sich da von allen Seiten die Wirkungsweise und gegenseitige Beeinträchtigung der Verbindung zwischen dem Wechselrichter und den Anwendungen dtuGateway, Hoymiles S-Miles Home Anbindung und jetzt auch noch everHome.

Aber wie geschrieben, läuft jetzt mit ein paar Aussetzern alles irgendwie auch parallel.

Gruß
Frank

Frankyistda, 2024-10-03:
@ohAnd Nochmal ich!

Jetzt zu den MQTT Daten. das wäre mir natürlich die liebere Lösung gewesen, die Daten für everHome bei Deiner dtuGateway via MQTT abzugreifen.

Ich habe gerade bei everHome mal nachgefragt, ob sie eine Schnittstelle zu dtuGateway schaffen wollen, die Antwort lautete eher nicht. Sie wollen sich halt nicht zu sehr verzetteln. Kann ich verstehen.

Mein Ansatz war jetzt, Deinen Code entsprechend hier am PC abzuändern um die MQTTs openDTU-like an everHome weiterzugeben, dafür gibt es ja schon eine Schnittstelle.

Ich vermute mal, die Änderungen müssten in dtuGateway.ino vorgenommen werden, so dass z.B. statt dtuGateway_12345678/grid/U dann dtuGateway_12345678/0/voltage wird.

Bei aller Vorsicht, ich habe da überhaupt keine tiefere Expertise in diesen Sachen. Meine beruflichen Programmierzeiten mir FORTRAN sind schon lange vorbei und in C bin ich eher schwach aufgestellt, von C++ ganz zu schweigen!

Trotzdem juckt es mir natürlich in den Fingern, die Nuss zu knacken :-)

Leider scheitere ich schon dabei, Deinen Original-Code zu kompilieren und zu binden.

Wenn ich das im Visual Studio Code mit dem PlatformIO Add-On (ohne mich in GitHub einzuloggen) versuche zu kompilieren und zu binden, dann kommt der Fehler:

ValueError: too many values to unpack (expected 3):
  File "C:\Users\tarno\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 167:
    env.SConscript(env.GetExtraScripts("pre"), exports="env")
  File "C:\Users\tarno\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\tarno\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\tmp\tmp2\version_inc.py", line 39:
    major, minor, patch = map(int, version_string.split('.'))

Da steh' ich nun, ich armer Tor, Und bin so klug als wie zuvor! :-)

Was mach ich falsch?

Oder hast Du evtl. die Zeit und Muße, eine openDTU-like MQTT Ausgabe mit in Dein Programm aufzunehmen? (ich frag' für einen Freund...)

Gruß und schönen Feiertag!
Frank

tengel07, 2024-10-03:
@Frankyistda 
Hallo Frank,

Laut everhome Webseite kannst Du Deinen hoymiles-Wechselrichter direkt in der everhome app einrichten, ohne das Du die DTU-Lösung benötigst.

VG und schönen (Rest)-Feiertag.
Torsten

Frankyistda, 2024-10-03:
@tengel07 

> Ich habe bei gleichzeitiger Anbindung des Hoymiles Wechselrichters an dtuGateway und der everHome Cloud via App (ich denke Du hast Recht, der ecoTracker hat damit nichts zu tun) festgestellt, dass dtuGateway oft "Offline" meldet.

Hallo Torsten,

ja, das wusste ich schon, hätte aber gerne die dtuGateway Lösung. Das Bessere ist ja bekanntlich der Feind des Guten ;-)

Gruß und schönen Feiertag
Frank

Frankyistda, 2024-10-03:
@ohAnd 

Zum oben erwähnten Fehler:

> ValueError: too many values to unpack (expected 3):
> File "C:\Users\tarno.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 167:
> env.SConscript(env.GetExtraScripts("pre"), exports="env")

Ich habe jetzt in der platformio.ini die beiden Einträge:
**extra_scripts = pre:version_inc.py**
auskommentiert.

So läßt sich das Ganze zumindest einmal übersetzen und zu einer firmware.bin binden (die allerdings 2480 byte größer ist als Deine dtuGateway_ESP32_release_2.0.0010.bin... ).

Ist das Deiner Meinung nach zulässig, Andreas?

Oder werde ich beim Aufspielen auf die esp32 scheitern?

ohAnd, 2024-10-09:
Hi @Frankyistda ,

sorry war offline... 
Wegen dem version script siehe hier -> https://github.com/ohAnd/dtuGateway?tab=readme-ov-file#build-environment

Wenn du durchcomplieren kannst und das richtige file erwischt, dann flashen und schauen, was passiert ;-)
Im Allgemeinen kannst du den ESP nicht so einfach "kaputt" flashen.

btw. 1: eine openDtu like MQTT Kommunikation noch zusätzlich einzubauen, würde ich mal ähnlich wie everHome beantworten ... "nicht zu sehr verzetteln" ;-) - daher erstmal aus meiner Sicht für das Projekt nicht sinnvoll

btw. 2: wie oben schon erwähnt finde ich das MQTT Autodiscovery Protokoll von (?) HomeAssistant ganz charmant, wird in meinem eigenem Setup auch problemlos von openHab erkannt - d.h. ein neues System bekommt "nur" den MQTT broker gesagt und pusht dann seine Config-Daten inkl. der Links (zu den Topics) wo welche Daten gelesen/ gesetzt werden können automatisch - verarbeitende Systeme wie HA und openHAB und sicher auch weitere binden dann diese Devices automatisch ein... - keine device spezifische Schnittstelle...

Gruss

closed 2024-12-30 by ohAnd