ohAnd/dtuGateway #59: MQTT Publish "Inverter power limit Set" as number

pitrade, 2024-10-19:
Problem:
MQTT `/inverter/PowerLimit_Set` becomes `sensor.()_inverter_power_limit_set` instead of `number.()_inverter_power_limit_set` HomeAssistant and is not editable with UI or automations. 

Suggestion:
Publish as number like [OpenDTU MqttHandleHass](https://github.com/tbnobody/OpenDTU/blob/master/src/MqttHandleHass.cpp#L203)

ohAnd, 2024-10-20:
@pitrade 

you're fully right - this point was never tested (and at least for me not used ;-) )

currently the set value is only a sensor - and in HA it will be only a display but no usable switch/ dimmer/ controller

I will check the needed changes to get a full integration for HA and other frameworks that using MQTT autodiscovery/ config

thanks for the hint!

pitrade, 2024-10-26:
@ohAnd works fine. Thank you!

pitrade, 2024-10-26:
There is a minimum value of 2%, so you can't set the limit to 0% when you won't discharge the battery. At least for me, a 0% minimum value would be better because my algorithm only sets the inverter limit. Then the inverter is turned on/off depending of the limit value.

Also an on/off button would be great. I currently use a custom Hoymiles integration for this but it is unstable. Should I create a new issue for this feature request?

ohAnd, 2024-10-27:
Hi @pitrade ,

yep, would be nice... (creating a new feature, due to other topic than setting the PowerLimit via MQTT and HA).
Then there is good point to extend the discussion around deactivating or things like that.

thx

closed 2024-10-27 by ohAnd