ohAnd/dtuGateway #61: Turning the inverter on and off

pitrade, 2024-11-07:
When turned on, even with the limit set to 0%, the inverter still draws about 8 W from my Battery. Therefore I recommend the feature to turn the device on and off with the web UI and also via MQTT with e.g. HomeAssistant.

ohAnd, 2024-11-26:
Hi @pitrade ,

it is now implemented - please check the new snapshot for your needs.
Not all input methods are already checked, so please tell what kind of setup is running on your site/what situation you were able to check to get a "hook" on the test list ;-)

BR

pitrade, 2024-11-27:
My Setup is dtuGateway on ESP32 and HomeAssitant with MQTT. I tested with the latest snapshot but didn't get control elements to turn the inverter on/off. Neither in HomeAssitant nor in the web interface. 
[image]
[image]

pitrade, 2024-11-28:
Turning the inverter OFF/ON by setting the powerlimit either to 0% or any value above works perfectly. Thank you a lot! Feel free to contact me for further beta testing in a ESP32+HomeAssistant environment.

ohAnd, 2024-12-10:
Hi @pitrade ,

I've implemented some extensions regarding this topic ... due to some playing/ testings around with some corner cases... for inverter on/off and back.
I didn't found something easy to get the current state for inverter on/off on dtu side. So I'm using now a cyclic polling of the current warnings of the dtu - because there is a specific entry in the list, when the "Inverter is remote off".

1. I also added a filter for changes of inverter on/off that prevents turn to off and on again within 60 seconds (and also vis-vis). (60 seconds for the testing phase - have to find best needed value)
2. the local inverter state off (in dtuGateway) will be checked with the warning (described above).
3. with the need to get the warnings - there was a new feature born for the webpage:
[image]
see also readme extension... - _not all error codes are already seen on my side, if you get unknown warnings please note the shown error code - and if possible check with the s-miles app and direct connection to the dtu the current readable warnings - so that I can extend the lookup table in code_

So, now to your offer ;-)

Please check with different use cases the points above. Especially the stability of turning inverter on/off and the state recognition...

Thx

pitrade, 2024-12-11:
Hi @ohAnd 

I updated to the latest snaphot and keep it running. My Home Assistant automatically sets the inverter limit depending on household consumption, battery SoC and grid electricity price.

pitrade, 2024-12-11:
[image]

pitrade, 2024-12-12:
This night the inverter turned on for no reason. The inverter was turned off with an additional script that triggers, when the voltage or temperature drops below a threshold.
[image]
[image]
[image]

ohAnd, 2024-12-12:
hi @pitrade ,

--------
at first to the error codes:

- error code 210 on both days at nearly the same time (00:00:58 <-> 00:00:53) - there is no second timestamp - means warning is active and there are no data (currently both data = 0) - means no additonal e.g. voltage value is given => that should belong to "[PV2 no input]" - as an assupmtion too.
- error code 218 almost the same but 5 min later (1st 00:05:58 - 2nd 00:05:53) - active warning at the time of screenshot, but with addtional data1 - this is 220 / 10 => 22.0 V and should be the min voltage level for the input PV2; data0 prints then the seen voltage level at the input (small bug here - with wrong interpreting 0 here) - so we cann assume here "[PV2 input undervoltage]"
- error code 124 you got on 11.12. at 9:45 and on 12.12. 01:07 - with your controlling by HA script - was the inverter shutted down at this time - means PowerLimitSet = 0? - so can we assume this belongs to the message "[Shut down by remote control]"? (interessting - on my side I got 8316 for "[Inverter remote off - active]")
- error code 16508 - from 1st day - was active between 00:01:22 and 09:05:34 - can we assume that during this time the inverter was in state off (by given PowerlimitSet = 0)? (on my side I got 28796 for an "[Inverter was remote off - period]")

--------
second - the "turned on for no reason" :

from code perspective i can currently not see a cause fur turning on the inverter if powerlimitset = 0 will be received or not changed on mqtt broker side - means it have to be a value > 0 given to send the inverter on request

maybe a side effect: 
could it be that the dtuGateway made a reboot at this time (power failure, local watchdog)?
currently the inverter state is internally On after booting and will be set to off with the given warning information (when the right error code is known - see above ;-) )
-> but this is only the local state ... with the "known" error code it will be then set to off with the first receiving of this code.

at this time it is diffcuilt to find the problem - it has to be supervised to get more infos of the surrounding circumstances
(with snapshot before 2.1.0066/2.1.0067- the problem was not given?)

pitrade, 2024-12-18:
@ohAnd I agree with your assumptions regarding the warnings. I got a new one today. Error code 72 with data0: 5051 and data1: 5020 should be related to the power reduction on high grid frequencies defined in [VDE-AR-N 4105](https://www.vde.com/resource/blob/2351688/daa3871967e7b2019f39318d1a11ef33/vde-anwendungsregel-anschluss-und-betrieb-von-erzeugungsanlagen-am-niederspannungsnetz-e-vde-ar-n-4105-2024-10-data.pdf). I suggest to add a sensor for the grid frequency which the inverter already monitors. 
[image]

The issue that the inverter turns on at midnight still persists and this did not happen with previous versions. Exactly at 00:00 the inverter turns on with a production that matches the last active limit. The web interface shows a limit of 0% but a production >> 0 W and the DTU state shows "dtu rebooting" as you suspected. You have to set the limit to >0% and back to 0% after the dtu reboot to turn the inverter off again
[image]

ohAnd, 2024-12-19:
Hi @pitrade ,

regarding the warnings I made some refoctoring in the recent snapshots - I also integrated the frequency warning - including the publication of the grid frequency to HA. Please check.

To the reactivate thing: which snapshot you are currently using with this wrong bahavior? (with the assupmtion that the snapshot n-1 worked for you) I will check then.
maybe side effect: Do you have cloud pause activated? - means 0,15,30,45 will be a short active disconnect to the dtu so that uploading of data is on dtu side not disturbed...
That it's happen exactly at 0:00 ... has to be investigated. ;-)

In one of [the recent snapshots](https://github.com/ohAnd/dtuGateway/commits/develop/?since=2024-12-01&until=2024-12-19) I've also added the real check of inverter status, means when the warning is active and the local inverter state is ON then it will set according to the feedback of dtu and vice versa. That should prevent undefined states locally. Means if there is still the powerlimitset of 0 given and the local state says inverter is off, but the dtu shows the warning "inverter is on" then the difference leads to a active set inverter off command to the dtu.
(could be not the root cause, but should be limit unexpected behaviours)

So, give it a try!

closed 2025-03-30 by ohAnd