ohAnd/dtuGateway #62: No connection to HMS-800W-2T possible

Maxinger15, 2024-12-24:
Hi,
my dtuGateway won't connect to my HMS.

I flashed the dtuGateway to my ESP and configured it.
I have also set up my HMS (connected it to my WIFI, added it via the S-Miles App to the Cloud) and when using the S-Miles app everything works fine. I can access my HMS over his own Wifi or over the Cloud.
But my dtuGateway won't connect to my HMS.

I tryed the latest release of dtuGateway (2.0.0013) and the current snapshot (2.1.0073).
I also updatet the HMS from Version 0.01.11 to 0.02.01.

Does anyone have an idea what could be the problem?

I get the following logs:

DTUinterface:	 getDataUpdate - ERROR - not connected to DTU!
>>>>> 05s task - state --> local: 20.02.1611 - 01:06:25 --- NTP: 13:34:55 ---> dtuConnState: 0
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 112110 ms - current conn state: 0
DTUinterface:	 dtuLoop - try to connect ... short: 5 - long: 0
DTUinterface:	 client not connected with DTU! try to connect (server: 192.168.1.49 - port: 10081) ...
>>>>> 05s task - state --> local: 20.02.1611 - 01:06:30 --- NTP: 13:35:00 ---> dtuConnState: 3
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 112122 ms - current conn state: 3
>>>>> 05s task - state --> local: 20.02.1611 - 01:06:35 --- NTP: 13:35:05 ---> dtuConnState: 3
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 112211 ms - current conn state: 3
>>>>> 05s task - state --> local: 20.02.1611 - 01:06:40 --- NTP: 13:35:10 ---> dtuConnState: 3
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 120000 ms - current conn state: 3
>>>>> 05s task - state --> local: 20.02.1611 - 01:06:45 --- NTP: 13:35:15 ---> dtuConnState: 3
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 120012 ms - current conn state: 3
>>>>> 05s task - state --> local: 20.02.1611 - 01:06:50 --- NTP: 13:35:20 ---> dtuConnState: 3
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 120101 ms - current conn state: 3
>>>>> 05s task - state --> local: 20.02.1611 - 01:06:55 --- NTP: 13:35:25 ---> dtuConnState: 3
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 120120 ms - current conn state: 3
DTUinterface:	 DTU Connection error: Connection aborted (-13)
DTUinterface:	 disconnected from DTU
>>>>> 32s task - state --> local: 20.02.1611 - 01:06:56 --- NTP: 13:35:26

DTUinterface:	 getDataUpdate - ERROR - not connected to DTU!
>>>>> 05s task - state --> local: 20.02.1611 - 01:07:00 --- NTP: 13:35:30 ---> dtuConnState: 0
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 120202 ms - current conn state: 0
DTUinterface:	 dtuLoop - PAUSE ... short: 6 - long: 0
>>>>> 05s task - state --> local: 20.02.1611 - 01:07:05 --- NTP: 13:35:35 ---> dtuConnState: 0
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 120221 ms - current conn state: 0
>>>>> 05s task - state --> local: 20.02.1611 - 01:07:10 --- NTP: 13:35:40 ---> dtuConnState: 0
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 121010 ms - current conn state: 0
>>>>> 05s task - state --> local: 20.02.1611 - 01:07:15 --- NTP: 13:35:45 ---> dtuConnState: 0
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 121022 ms - current conn state: 0
>>>>> 05s task - state --> local: 20.02.1611 - 01:07:20 --- NTP: 13:35:50 ---> dtuConnState: 0
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 121111 ms - current conn state: 0
>>>>> 05s task - state --> local: 20.02.1611 - 01:07:25 --- NTP: 13:35:55 ---> dtuConnState: 0
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 121200 ms - current conn state: 0
>>>>> 32s task - state --> local: 20.02.1611 - 01:07:28 --- NTP: 13:35:58

DTUinterface:	 getDataUpdate - ERROR - not connected to DTU!
>>>>> 05s task - state --> local: 20.02.1611 - 01:07:30 --- NTP: 13:36:00 ---> dtuConnState: 0
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 121212 ms - current conn state: 0
>>>>> 05s task - state --> local: 20.02.1611 - 01:07:35 --- NTP: 13:36:05 ---> dtuConnState: 0
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 122001 ms - current conn state: 0
>>>>> 05s task - state --> local: 20.02.1611 - 01:07:40 --- NTP: 13:36:10 ---> dtuConnState: 0
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 122020 ms - current conn state: 0
>>>>> 05s task - state --> local: 20.02.1611 - 01:07:45 --- NTP: 13:36:15 ---> dtuConnState: 0
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 122102 ms - current conn state: 0
>>>>> 05s task - state --> local: 20.02.1611 - 01:07:50 --- NTP: 13:36:20 ---> dtuConnState: 0
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 122121 ms - current conn state: 0
>>>>> 05s task - state --> local: 20.02.1611 - 01:07:55 --- NTP: 13:36:25 ---> dtuConnState: 0
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 122210 ms - current conn state: 0
>>>>> 05s task - state --> local: 20.02.1611 - 01:08:00 --- NTP: 13:36:30 ---> dtuConnState: 0
>>>>> 32s task - state --> local: 20.02.1611 - 01:08:00 --- NTP: 13:36:30

DTUinterface:	 getDataUpdate - ERROR - not connected to DTU!
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 122222 ms - current conn state: 0
>>>>> 05s task - state --> local: 20.02.1611 - 01:08:05 --- NTP: 13:36:35 ---> dtuConnState: 0
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 200011 ms - current conn state: 0
DTUinterface:	 dtuLoop - try to connect ... short: 1 - long: 0
DTUinterface:	 client not connected with DTU! try to connect (server: 192.168.1.49 - port: 10081) ...
>>>>> 05s task - state --> local: 20.02.1611 - 01:08:10 --- NTP: 13:36:40 ---> dtuConnState: 3
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 200100 ms - current conn state: 3
>>>>> 05s task - state --> local: 20.02.1611 - 01:08:15 --- NTP: 13:36:45 ---> dtuConnState: 3
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 200112 ms - current conn state: 3
>>>>> 05s task - state --> local: 20.02.1611 - 01:08:20 --- NTP: 13:36:50 ---> dtuConnState: 3
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 200201 ms - current conn state: 3
>>>>> 05s task - state --> local: 20.02.1611 - 01:08:25 --- NTP: 13:36:55 ---> dtuConnState: 3
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 200220 ms - current conn state: 3
>>>>> 05s task - state --> local: 20.02.1611 - 01:08:30 --- NTP: 13:37:00 ---> dtuConnState: 3
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 201002 ms - current conn state: 3
>>>>> 32s task - state --> local: 20.02.1611 - 01:08:32 --- NTP: 13:37:02

DTUinterface:	 getDataUpdate - ERROR - not connected to DTU!
>>>>> 05s task - state --> local: 20.02.1611 - 01:08:35 --- NTP: 13:37:05 ---> dtuConnState: 3
DTUinterface:	 setOverallOnlineOfflineState - timeout - reset online offline state - difference: 201021 ms - current conn state: 3
DTUinterface:	 DTU Connection error: Connection aborted (-13)
DTUinterface:	 disconnected from DTU

ohAnd, 2024-12-28:
Hi @Maxinger15 ,

the connection should be directly possible if the dtu (HMS) has the needed configuration to be in your local wifi and the dtuGateway is within the same network. 
Can you ping your dtu (HMS) at 192.168.1.49 (seems your configured IP of the dtu) from within your network?

(btw: Do you're using an ESP32 oder 8266?)

Generally there is no specific setting needed from my perspective to get the basic connection from dtuGateway to dtu.

So, please check your network environment...

ohAnd, 2025-02-03:
not answered for 6 weeks

closed 2025-02-03 by ohAnd