ohAnd/dtuGateway #63: Web Interface lässt sich nicht mehr aufrufen

checkersky, 2025-01-22:
Hallo, ich habe das Problem das ich in der aktuellen Snapshots das Webinterface nicht mehr aufrufen kann.
Egal mit welchem Browser ( safari, Firefox, Edge ) ich bekomme nur eine leere Seute angezeigt.

Ich kann die Seite nur 1x aufrufen zum Konfigurieren danach geht es nicht mehr.

Die Letzte Version wo ich das Webinterface aufrufen kann ist die  v2.0.0013 bei allen Snapshots danach geht es nicht mehr.

ohAnd, 2025-02-03:
Hi @checkersky,

sorry ganz übersehen ...
Wichtigste Frage auf welchem Controller läuft bei Dir dtuGateway?

Spoiler: ESP8266 wird langsam zu knapp und ist in den späteren Snapshots auf jeden Fall "murks".
Mit dem ESP32 sehe ich bei mir keine Themen in verschiedenen Varianten. Auch Windows Chrome, Android Chrome, IOS Safari/ Chrome...

Daher bitte die Umgebung ein bisschen eingrenzen, um hier einen möglichen Fehler zu finden.

checkersky, 2025-02-03:
Hi, ja sorry ganz vergessen es ist noch der ESP8266.

checkersky, 2025-02-10:
Ich habe nun alle Browser Probiert von Safari IOS bis Firefox,Chrome und Edge unter Windows 11.
Bei allen das gleiche es passiert nichts nur eine weiße seite

oxidizer2001, 2025-02-11:
Hier das gleiche beim Latest snapshot v2.1.0073_develop. Ich habe das AZDelivery NodeMCU V2 WiFi Amica ESP8266 ESP-12F.

ohAnd, 2025-02-19:
Hi zusammen,

wie oben beschrieben, ESP8266 ist mit den Erweiterungen bzgl. Fehlerlog lesen und verarbeiten nicht mehr ausreichend. (mglw. mit viel Aufwand vielleicht) 

Komme aktuell nicht allzu schnell dazu, die Themen auseinander zu ziehen.
Ziel, sofern das sinnvoll darstellbar ist, den Part nur für ESP32 aktiv zu halten, so dass 8266 noch ein paar Iterationen unterstützt werden kann.

So far. 😉

checkersky, 2025-02-27:
Hi, nach Umstellung auf den ESP32 funktioniert alles wieder =)

closed 2025-02-27 by checkersky