ohAnd/dtuGateway #66: Add possibility for port settings for the dashboard

OE9SAU, 2025-03-06:
Hello Andreas
this project blows my mind. It does what it needs to do. Well done.
But i still have a feature request, to add the possibility for any kind of port settings. example dashboard

Regards Martin

ohAnd, 2025-03-10:
Hi @OE9SAU ,

thanks for appreciating!

Can you please detailing a little more what you mean with the both points?
1. port of the dashboard means other than port 80? or ports for the different binding connections?
2. user and password for what ;-) - for wifi connection and mqtt it's already there ...

thx

OE9SAU, 2025-03-10:
Hello @ohAnd,
for 1, freely definable webport, instead of the standard 80
for 2, User and password to access the settings

Greetings Martin

ohAnd, 2025-03-11:
@OE9SAU can you please explain / describe for both - what it the usecase behind? - and so, what is the real user need ... 

to 1. why it should be make sense to have this changable
- from a user perspective I need to know that it has a specific port instead of only putting the ip/ domain name in the browser ...
- this device has a single service provided with an own dedicated IP ... (means no double used ports or something in sharing with other services)
... but maybe there is soemthing more I don't see....

to 2. for this very small application and the kind of displaying data it's nothing for a dummy user - so against whom the settings data should be protected inner a local environment with a non ssl webserver?

gimme some hints for my understanding why it is a useful enhancement - so the spended effort is valuable

OE9SAU, 2025-03-12:
Good morning
In fact, my requirements are rather specific. This is because I would like to make the dtu gateway accessible externally, not just internally. Hence the password and user name (point 2), at least for the settings, that noboday can change something.

Regarding 1: We the ÖVSV (Österreichischer Versuchssenderverband) operate a radio relay station at one location. This is also the place where the dtu will be placed. This location has a fixed IP address in the WAN. However, several services run on different IPs in the LAN. But from the outside, only one IP is available, so that not everything has to be mapped with Dyndns, etc. – the idea behind the web port.

As I said, the dtu is brilliant and serves its purpose. Everything else is just nice to have. Regards Martin

ohAnd, 2025-03-30:
Hi @OE9SAU ,

the webserver port change is now inmplemented - at first in this specific feature branch 

Please check if it fulfills your needs. After positive testing I will pull to the dev for the next snapshot/ release.

The firmware is attached here:

[dtuGateway_issue66_2.2.xx.bin.zip](https://github.com/user-attachments/files/19526667/dtuGateway_issue66_2.2.xx.bin.zip)

OE9SAU, 2025-03-31:
Hello @ohAnd,
will check asap and let you know.
Regards Martin

ohAnd, 2025-07-08:
> What about the password for the settings page? What's the latest status there?

check open questions here -> #71