ohAnd/dtuGateway #71: user and password for changing settings over the dasboard

ohAnd, 2025-03-30:
based on #66 

... In fact, my requirements are rather specific. This is because I would like to make the dtu gateway accessible externally, not just internally. Hence the password and user name (point 2), at least for the settings, that noboday can change something. ...

ohAnd, 2025-03-30:
@OE9SAU I've splitted #66 to discuss here the further details for this idea...

For this goal there are several ways possible - with a lot of effort or with less or more less ...

ideas:
1. full refactoring to https and using also encrypted data handling for user config data and so on (a lot of work)
2. as another idea: with http and user session handling (medium effort)
      - means unsecure transport of user / password
      - and only with that the user config settings will be provided via api and are then changable
      - need session keys, handling of that and further stuff
3. pragmatic way, if it's fitting to your situation (small effort)     
      - blocking/ unblocking the settings view/ edit by a direct serial cmd
      - means change setting and/ or displaying it over web / and providing via API will be disabled
      - assumes the config has not to be changed from remote ...

let's discuss a bit to find the right solution - based on your specific needs
and let's start with number 3 ;-)

btw. interestng kind of application for the duGateway

OE9SAU, 2025-07-08:
Hi Andreas,
number 3 pragmatic way, is still enough for my needs. blocks unauthorized access to advanced settings.
Many  thanks for implementing. Regards Martin

ohAnd, 2025-07-10:
@OE9SAU please check last develop for that feature

OE9SAU, 2025-07-14:
Good morning Andreas
Protecting the config via serial commands works great and currently meets my needs 100%. Thanks for that.
Martin

commit 6a6a95e9 (2025-07-10): Add settings protection feature

- Introduced `protectSettings` boolean in UserConfig to prevent changes via web interface.
- Enhanced web interface to display protection status and prevent unauthorized changes.
- Updated README to document the new settings protection command. - fix #71 user and password for changing settings over the dasboard