ohAnd/dtuGateway #76: TFT-Display doesn't change back to normal after Inverter switch off via App

zigit780, 2025-05-08:
I was doing a switchoff from the inverter with the Hoymile app and then a switch on.

The TFT-display shows inverter off correctly when it was switched off thats fine.
But after switch on the TFT-display doesn't update the display back to normal with power-display.
In the webpage I see the hint that there was a switch off and the rest shows the normal things thats also ok.

But the TFT doesn't go back to normal. If this is wanted, I would prefer to go back to normal with mayby extra symbol that there was a switch off. And then a possibility for delete this message in the webpage.

What would be also good to have the possibility to restart the DTUgateway-Module. 
Or is there any but don't find it?

Inverter: HMS-800W-2T
FW: V.1.03.09
HW: H00.04.00

ohAnd, 2025-05-20:
That should  normally not happen - is that reproducable on your site?

Means if inverter is switched on again the normal display should appear.

For restarting - it's currently only directly possible with serial connection per cmd.
With a 'trick' you can pushing it also with web ui - every change via the advanced config triggering a reset ... ;-)

zigit780, 2025-05-30:
Yes is reproducable.

zigit780, 2025-06-19:
after >30 min not switching back, then I changed brightness value via advanced config then it come back (like you said)