ohAnd/dtuGateway #79: ValueError: too many values to unpack (expected 3)

peter-tue, 2025-05-17:
With the current snapshot with latest build version: 2.2.1.0.0 I get an error in VSC/PIO for ESP32:

`Processing esp32 (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
get current version: 2.2.1.0.0
 from D:\ESP\PVA/include/version.h
ValueError: too many values to unpack (expected 3):
got buildnumber to use: ->"localDev"<-
  File "C:\Users\p\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 167:
    env.SConscript(env.GetExtraScripts("pre"), exports="env")
  File "C:\Users\p\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\p\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\ESP\PVA\version_inc.py", line 39:
    major, minor, patch = map(int, version_string.split('.'))
============================================ [FAILED] Took 1.94 seconds ======`

I had already added the still missing 'buildnumber.txt' withe the content "localDev".

How can I get it to compile ?

Thanks in advance.

ohAnd, 2025-05-19:
this line from you log
`get current version: major.minor.patch from D:\ESP\HMS800W/include/version.h` 
shows (I have to assume) that you have in your version.h this 
```c
#define VERSION "major.minor.patch_localDev"
#define BUILDTIME "23.04.2025 - 07:19:17"
#define BUILDTIMESTAMP "1745385557"
```
but should be something like this
```c
#define VERSION "2.2.160_localDev"
#define BUILDTIME "23.04.2025 - 07:19:17"
#define BUILDTIMESTAMP "1745385557"
```

check this....

peter-tue, 2025-05-19:
I now started new from the current snapshot 2.2.0079.zip (without adding the missing buildnumber.txt) with this version.h:

#define VERSION "2.2.160_localDev"
#define BUILDTIME "23.04.2025 - 07:19:17"
#define BUILDTIMESTAMP "1745385557"

and get :

```
Verbose mode can be enabled via `-v, --verbose` option
get current version: 2.2.160_localDev
 from D:\ESP\HMS800W/include/version.h
ERROR: buildnumber file not found. For automatic versioning there is a file called ../include/buildnumber.txt expected. With the content "localDev" or versionnumber e.g. "1.0.0" in first line. (File is blocked by .gitignore for GitHub actions to run.)
NameError: name 'build_string' is not defined:
  File "C:\Users\p\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 167:
    env.SConscript(env.GetExtraScripts("pre"), exports="env")
  File "C:\Users\p\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\p\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\ESP\HMS800W\version_inc.py", line 41:
    if build_string != "localDev":
========================== [FAILED] Took 4.83 seconds ====
```

The content of the version.h remains unchanged.
When I add a file buildnumber.txt withe teh content "localDev" in the first line the version. get changed to:

#define VERSION "2.2."localDev""
#define BUILDTIME "19.05.2025 - 18:02:52"
#define BUILDTIMESTAMP "1747670572"

and I get this error:

```
Verbose mode can be enabled via `-v, --verbose` option
get current version: 2.2.localDev
 from D:\ESP\HMS800W/include/version.h
got buildnumber to use: ->"localDev"<-
ValueError: invalid literal for int() with base 10: 'localDev\n':
  File "C:\Users\p\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 167:
    env.SConscript(env.GetExtraScripts("pre"), exports="env")
  File "C:\Users\p\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\p\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\ESP\HMS800W\version_inc.py", line 39:
    major, minor, patch = map(int, version_string.split('.'))
=================================================== [FAILED] Took 3.41 seconds ===
```

It seems I'm generating the wrong buildnumber.txt. Could you please add a correct buildnumber.txt to this project?

Thanks, Peter

ohAnd, 2025-05-19:
Try as mentioned... 2.2.0_localdev

According to major minor patch + the specific suffix

peter-tue, 2025-05-20:
Thanks again for your help and that great work for that project.

The 'trick ' was to have the text >localDev< in the buildnumber.txt file without quotes.

with that content the version.h gets automatically changed to:

`#define VERSION "2.2.162_localDev"
#define BUILDTIME "20.05.2025 - 07:59:07"
#define BUILDTIMESTAMP "1747720747"`

and the project could be compiled 👍

closed 2025-05-20 by ohAnd