ohAnd/dtuGateway #80: ESP32 - S3 don´t boot

AmshaSharam, 2025-05-19:
i try flashing an ESP32-s3 with adicional usb port but afterflashing keeps booting , any help.
Flashed with Flash Download tool.
if i flash with opendtu firmware on the web flasher its works.

ohAnd, 2025-05-19:
Hi @AmshaSharam ,

I guess way to less information ... ;-) 
check the readme for first time flashing (especially for ESP32 use the right additional files) and go step by step with this instructions - and document here incl. maybe logfiles during flashing - to have some valuable info what can be discussed.

AmshaSharam, 2025-05-19:
after reading openDtu , looks like for ESP32 - S3 there are files for that chips, they also have xxxx.factory.bin for use in the first time , and it works on my esp.
I got an ESP32-s3 Wroom-1 N16R8
After flashing like the post refered on instalation intrustions. the flash don,t give any errors, but the esp keep flashing , and on terminal i can´t connect because keeps booting.

ohAnd, 2025-05-20:
can you explain in detail which steps you are running through and which files (from where) you use to flash for the first time

AmshaSharam, 2025-05-24:
Can you bild please a version for the esp32- S3 ?

AmshaSharam, 2025-05-25:
Thank you very much
Unfortunately it didn't work.
I've been reading about it, I think you have to provide me with the "bootloader.bin" files, etc., for this board, the settings for the Flash tool also change for this board.

[image]
[image]
[image]

board: esp32-s3-devkitc-1 - n16r8v

ohAnd, 2025-05-25:
oh forgot to mention - check https://github.com/ohAnd/dtuGateway/tree/develop/doc/esp32_S3_factoryFlash

AmshaSharam, 2025-05-25:
Now its working but its stays on this screen: 

[image]

i use usb port to flash 

[image]

my flash settings:
1º selected: 

[image]

2º:

[image]

i reset and etc.

Try terminal serial port uart , usb don't show data getting this:

[image]
  
it's on good way, many thanks

ohAnd, 2025-05-26:
;-) as said not supported - because not testable on my side ...

let's try further

your pre last post:
Interesting that you get the website - do you could also configure your local wifi? -> additionally check here - seems similiar point https://github.com/ohAnd/dtuGateway/discussions/78#discussioncomment-13209313

last post: 
```
[ 435][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
Guru Meditation Error: Core 1 panic'ed (StoreProhibited). Exception was unhandled.
```
that's interesting ... when this happend? - after every startup or only every second startup - in the state that the dtuGateway is still in factory mode? (asked with intention)

I've checked shortly the needed IOs which are used in this project with the result the common ESP32 S3 should be compatible with the given setup... 

So please check your board and schematic for uncompatbility to the given setup ... to find the cause why you getting here `Invalid pin selected`
used GPIOs are 
=> https://github.com/ohAnd/dtuGateway?tab=readme-ov-file#hardware
and GPIO 2 for a LED (blinking info)

ohAnd, 2025-06-07:
@AmshaSharam is it now running ?

If yes, maybe you can describe for a short what was maybe the problem?
Then I can take it over to the readme.

btw. do you also using a display (which one) and it also working and with which pinning?

AmshaSharam, 2025-06-11:
Sorry, i´m busy with work. I find that my board is ESP32-S3 with 16 MB Flash 8 MB PSRAM + usb, i try to compile with settings for this board , but didn´t work. If i use [opendtu-generic_esp32s3_usb.factory.bin] it works, im trying to find the right settings on platformio to work.

AmshaSharam, 2025-06-14:
hi 
Finally i got it, im now testing with ESP32-S3-WROOM-1-N16R8 connected to HMS-1000W-2T.
how i do:
On platformio.ini :
-----------------------------------------------------------------
; Flash: 16MB QD, PSRAM: 8MB OT
[env:esp32-s3-devkitc-1-n16r8]
platform = espressif32
board = esp32-s3-devkitc-1-n16r8v
framework = arduino

board_build.arduino.memory_type = qio_opi
board_build.flash_mode = qio
board_build.psram_type = opi
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.partitions = default_16MB.csv
board_build.extra_flags = 
  -DBOARD_HAS_PSRAM

monitor_speed = 115200
monitor_port = COM4
upload_port = COM4
upload_speed = 921600
lib_deps = 
	arduino-libraries/NTPClient @ ^3.2.1
	robtillaart/CRC @ ^1.0.2
	nanopb/Nanopb @ ^0.4.8
	gyverlibs/UnixTime @ ^1.1
	bblanchon/ArduinoJson @ ^7.0.0
	knolleary/PubSubClient @ ^2.8
	olikraus/U8g2 @ ^2.35.19
	bodmer/TFT_eSPI @ ^2.5.43
	ESP32Async/AsyncTCP @ 3.4.2
	ESP32Async/ESPAsyncWebServer @ 3.7.7
	
custom_nanopb_protos = 
	+<include/proto/AppGetHistPower.proto>
	+<include/proto/RealtimeDataNew.proto>
	+<include/proto/GetConfig.proto>
	+<include/proto/CommandPB.proto>
	+<include/proto/AlarmData.proto>
extra_scripts = pre:version_inc.py
monitor_filters = 
	esp32_exception_decoder
	default
	time
build_flags = 
	-DUSER_SETUP_LOADED=1
	-DGC9A01_DRIVER=1
	-DTFT_WIDTH=240
	-DTFT_HEIGHT=240
	-DTFT_MISO=19
	-DTFT_MOSI=23
	-DTFT_SCLK=18
	-DTFT_CS=15
	-DTFT_DC=2
	-DTFT_RST=-1
	-DLOAD_GLCD=1
	-DLOAD_FONT2=1
	-DLOAD_FONT4=1
	-DLOAD_FONT6=1
	-DLOAD_FONT7=1
	-DLOAD_FONT8=1
	-DLOAD_GFXFF=1
	-DSMOOTH_FONT=1
	-DSPI_FREQUENCY=27000000

than flashed on COM port.
-----------------------------------------------------------

Now i will connect the screen to see
its working

AmshaSharam, 2025-06-14:
Just change the title :)

[image]

ohAnd, 2025-06-29:
ok, thanks looks good... 
I've tried to bring this as additonal to build and see that your defined board not exist in platform.io...

I've build this also with devkitc-1 but additionally to support 16MB and PSRAM with your mentioned build-flags.

Can you please test the dev build?

Then it will be part of the standard builds fur further changes...

Title thing have to be another issue - so it can be automized by recognition of the type of the inverter... (so you don't need your "own" build fur further updates .. ;-) )

AmshaSharam, 2025-07-07:
hi, i try flash your bin file, didnt work, i flashed by UART and USB,  keeps booting, i use my bootloader.bin and etc, no lucky. Put this file on "platformio\platforms\espressif32\boards" 

[esp32-s3-devkitc-1-n16r8v.json](https://github.com/user-attachments/files/21109529/esp32-s3-devkitc-1-n16r8v.json)