ohAnd/dtuGateway #81: Warnings staying active in dtuGateway, but in Hoymiles App are no warnings

ohAnd, 2025-05-22:
So noch einmal von heute Mittag. Die Meldung ist und bleibt wie gestern Abend und heute Morgen.

[image]
[image]

_Originally posted by @OpenShorty in https://github.com/ohAnd/dtuGateway/discussions/78#discussioncomment-13218976_

Heini1985, 2025-06-27:
Hallo zusammen ich habe auch immer wieder eine Warnung.
[image]

[image]

Der Hoymules ist aber normal im Betrieb. In der App ist kein Fehler vorhanden.

Viele Grüße.

Heini1985

chrisr79, 2025-08-03:
Lt. diversen Foren ist der Alarm 72 ein: "Power grid over-frequency load reduction (FW) function enable".