ohAnd/dtuGateway #104: Openhab Item prefix Problem

Benni620, 2025-07-30:
**Describe the bug**
I use the latest fw version: 2.2.0096_develop. When I configure the OH interface, I set the prefix to "inverter" and I used the Item template from the docs. Then I got errors and only a few Items got values:

Number inverter_Temp "Inverter Temperature [%.1f °C]"
Number inverter_PowerLimit "Current Power Limit [%d %%]"
Number inverter_WifiRSSI "DTU Wi-Fi Signal [%d %%]"

The others were expected to be called:
Number inverterGrid_U "Grid Voltage [%.1f V]"
Number inverterGrid_I "Grid Current [%.2f A]"
Number inverterGrid_P "Grid Power [%.0f W]"
Number inverterPV_E_day "Daily Energy [%.3f kWh]"
...and so on...

**Expected behavior**
The underscore should be ther on ervery Item ;)

I hope that are enought informations, if you need more please ask.

And I have to say, what a cool job you're doing here!
Greetings

ohAnd, 2025-07-30:
Hi @Benni620 ,

thx.

And sorry, that is a "bug" in the readme after the redesign of the readme ... (see [here](https://github.com/ohAnd/dtuGateway/blob/main/readme_old.md#openhab-integration-configuration) the old one)

current readme
```
// Read power limit setting from openHAB
Number inverter_PowerLimitSet "Power Limit Set [%d %%]"

// Grid data
Number inverter_Grid_U "Grid Voltage [%.1f V]"
Number inverter_Grid_I "Grid Current [%.2f A]"
Number inverter_Grid_P "Grid Power [%.0f W]"
Number inverter_PV_E_day "Daily Energy [%.3f kWh]"
Number inverter_PV_E_total "Total Energy [%.3f kWh]"

// Panel 1 data  
Number inverter_PV1_U "PV1 Voltage [%.1f V]"
Number inverter_PV1_I "PV1 Current [%.2f A]"
Number inverter_PV1_P "PV1 Power [%.0f W]"
Number inverter_PV1_E_day "PV1 Daily Energy [%.3f kWh]"
Number inverter_PV1_E_total "PV1 Total Energy [%.3f kWh]"

// Panel 2 data
Number inverter_PV2_U "PV2 Voltage [%.1f V]"
Number inverter_PV2_I "PV2 Current [%.2f A]"
Number inverter_PV2_P "PV2 Power [%.0f W]"
Number inverter_PV2_E_day "PV2 Daily Energy [%.3f kWh]"
Number inverter_PV2_E_total "PV2 Total Energy [%.3f kWh]"

// Inverter status
Number inverter_Temp "Inverter Temperature [%.1f °C]"
Number inverter_PowerLimit "Current Power Limit [%d %%]"
Number inverter_WifiRSSI "DTU Wi-Fi Signal [%d %%]"
```

but in code is expected like this:
```
// Read power limit setting from openHAB
Number inverter_PowerLimitSet "Power Limit Set [%d %%]"

// Grid data
Number inverterGrid_U "Grid Voltage [%.1f V]"
Number inverterGrid_I "Grid Current [%.2f A]"
Number inverterGrid_P "Grid Power [%.0f W]"
Number inverterPV_E_day "Daily Energy [%.3f kWh]"
Number inverterPV_E_total "Total Energy [%.3f kWh]"

// Panel 1 data  
Number inverterPV1_U "PV1 Voltage [%.1f V]"
Number inverterPV1_I "PV1 Current [%.2f A]"
Number inverterPV1_P "PV1 Power [%.0f W]"
Number inverterPV1_E_day "PV1 Daily Energy [%.3f kWh]"
Number inverterPV1_E_total "PV1 Total Energy [%.3f kWh]"

// Panel 2 data
Number inverterPV2_U "PV2 Voltage [%.1f V]"
Number inverterPV2_I "PV2 Current [%.2f A]"
Number inverterPV2_P "PV2 Power [%.0f W]"
Number inverterPV2_E_day "PV2 Daily Energy [%.3f kWh]"
Number inverterPV2_E_total "PV2 Total Energy [%.3f kWh]"

// Inverter status
Number inverter_Temp "Inverter Temperature [%.1f °C]"
Number inverter_PowerLimit "Current Power Limit [%d %%]"
Number inverter_WifiRSSI "DTU Wi-Fi Signal [%d %%]"
```
(btw. you should see in openhab log, that there are requests to update items that are not exists - and there the names are mentioned as bove)

I will update the readme in that way.

PS: And yep... ;-) the readme version looks more consistent, but a change to the "right way" will be ful incompatible. The current implementation for that part is existing since start of the project...

ohAnd, 2025-08-03:
85fe2e36c01a0b31cffa2e8e299b208187a14ad2 fixes the doc

closed 2025-08-03 by ohAnd